ARMmbed/mbed-os #4972: Possibly non relevant check in tests-mbed_drivers-flashiap

LMESTM, 2017-08-24:
### Description
tests-mbed_drivers-flashiap | FlashIAP - program errors  test checks for several return codes and expects that flash_program_page will return error (-1) in case of "unaligned" address or "unaligned" page size. 

But as discussed in #4966 STM32F4 targets are able to program flash with 1 byte granularity, so from common understanding so far, the get_page_size shall return 1 (minimum program size of 1). Then considering that we can program 1 byte at a time, the above error check does not make sense.

LMESTM, 2017-08-24:
Note: I will propose a fix for that one

LMESTM, 2017-08-28:
Fix proposed in #4980

0xc0170, 2017-08-31:
cc @c1728p9

0xc0170, 2017-08-31:
> But as discussed in #4966 STM32F4 targets are able to program flash with 1 byte granularity, so from common understanding so far, the get_page_size shall return 1 (minimum program size of 1). Then considering that we can program 1 byte at a time, the above error check does not make sense.

As already asked somewhere else, is 1 byte actually reasonable to have (regarding performence) ? Even if the device is capalbe of this, when writing lets say image size of 1MB, the performance would be much lower compare to bigger pages sizes ? Is there any tradeoff ? What is the default number set in the ST drivers ?

LMESTM, 2017-08-31:
Reading again the API:
get_page_size is now the "minimum" programmable size ... but this looks also like the maximum that will be used because the programming API is flash_program_page which sounds like it will only be used to program one page at a time even though the STM32 hardware and HAL implementation does not have such restriction ...   how can offer both capabilities to users: writing either 1 byte or big chunks of data ?

LMESTM, 2017-09-13:
#4980 was merged - so this issue can be closed

closed 2017-09-13 by LMESTM