ARMmbed/mbed-os #4976: Slow tests builds with memmap script updates

c1728p9, 2017-08-25:
When I run `mbed test -t GCC_ARM -m K64F --compile` to do an incremental rebuild (with no changes) it takes nearly 2 minutes to link the first test. Before #4392, linking all tests completed in < 15 seconds.

c1728p9, 2017-08-25:
CC: @theotherjimmy @studavekar @MarceloSalazar

theotherjimmy, 2017-08-28:
@c1728p9 Definitely a bug. Here's some more data:
>My personal laptop specs:
i5-3320M @ 2.4 Ghz + 8GB DDR3 + 128GB SSD + Up to date Archlinux

> Testing Results:
Time to first link: about 20 sec.
link throughput: about 1 every 5 sec.

> My work machine:
Xeon E3-1240 v3 @ 3.8Ghz+ 32GB DDR3 + 256GB SSD + Up to date Archlinux

> Testing Results
Time to first link: 2 sec.
link throughput: 4 per second.

I suppose I did not notice a difference on my work machine. Sorry about that. I'll schedule looking into it.

0xc0170, 2017-08-30:
I rebased some of my branches to the latest master and facing this slowdown as well. This is the second bug that referenced PR introduced, they both affect testing (singletest and test).

MarceloSalazar, 2017-09-06:
That's weird, I'm not seeing the reported behavior. It takes me about 10 seconds to do an incremental rebuild of all mbed OS tests on my Macbook pro.

MarceloSalazar, 2017-09-11:
Is this still a problem? I'm curious to understand what's causing the delay, and if it's case, see could be re-worked/optimized.

theotherjimmy, 2017-09-11:
@MarceloSalazar, Yes this is still a problem. It seems to strongly depend on your setup, and which python version you are using. I have done some tests, and it looks like there are just some inefficiencies in the parsing code that may depend on L1/L2 cache size or something. For example, I was able to get a 25% speedup by moving [this line](https://github.com/ARMmbed/mbed-os/blob/master/tools/memap.py#L102) out of the for loop. I should have reviewed the code more closely to catch things like that, and I will in the future.

theotherjimmy, 2017-09-15:
It looks like it's scanning related, and that TESTS perform much worse because of this line section: https://github.com/ARMmbed/mbed-os/blob/master/tools/memap.py#L537-L554

PR #5125 (2017-09-21): Refactor memap for speed
Improves performance by 10-15x on my machine. See below for individual compiler comparisons.

Resolves #4976
Resolves #5124
Resolves #5127 

Todo:
 - [x] Parsing tests (let's never let this non-performance bug in again)
 - [x] Resolve the `Malformed input...` output from GCC
closed 2017-09-21 by theotherjimmy