ARMmbed/mbed-os #4978: RealTek8195AM can NOT initialize WiFi and got stock.

YiChunHung, 2017-08-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question|Bug
- Related issue: 

---------------------------------------------------------------
## Bug

**Target**
REALTEK8195AM

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

## Question

**How to?**
I am the newer to the realtek8195AM.
After following the installation of mbed-os on realtek8195AM and burn the WiFi example,
I use the `screen` to monitor the board.
I found that my board seems to stock on certain point.
I can't figure out what happened to it.
[image]

bulislaw, 2017-08-29:
Can you try any other example like blinky? https://github.com/ARMmbed/mbed-os-example-blinky please note that the board does not have an user LED so you must add external one or simply add printf in the loop and watch serial.

YiChunHung, 2017-08-29:
I did try the blinky example with adding printf in the loop.
It does not work. It got stuck at the same point.

bulislaw, 2017-08-29:
Try updating the firmware: https://www.amebaiot.com/en/change-dap-firmware/

YiChunHung, 2017-08-29:
I just update the firmware two days ago from the website you post.
I update the one name "DAP_FW_MBED_RTL8195AM_V0" and I all follow the steps described in the web.
I will try it again later, but I don't think it would work.

bulislaw, 2017-08-29:
@Archcady @tung7970 any ideas guys?

tung7970, 2017-08-29:
The blinky example should have worked. That's the basic HAL thing.

@YiChunHung can you paste the following details
1. mbedls output 
2. blinky build command

The latest DAPLink support for RTL8195AM is currently under review.
https://github.com/mbedmicro/DAPLink/pull/289

You can build one if you have the proper tool.

YiChunHung, 2017-08-29:
OK, I will try the latest one, and do some testing.
If not working, I will paste the information here.
Thanks for your help.

YiChunHung, 2017-08-29:
@tung7970 @bulislaw  Thanks for helping.
I solved the problem.
Online compiler seems not to work.
I used mbed-cli to compile the program and burn it to the board, it worked totally fine!!
By the way, I still pose the information I got. Probably it can help online compiler improve somehow.
1. mbedls output:
```
+-------------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
| platform_name     | platform_name_unique | mount_point      | serial_port           | target_id                                        | daplink_version |
+-------------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
| REALTEK_RTL8195AM | REALTEK_RTL8195AM[0] | /Volumes/DAPLINK | /dev/tty.usbmodem1412 | 46000000098a603b00000000000000000000000097969902 | 0241            |
+-------------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
```

2. Blinky build command

`mbed compile -t GCC_ARM -m REALTEK_RTL8195AM `
Here is the result
```
Building project mbed-os-example-blinky (REALTEK_RTL8195AM, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
Scan: FEATURE_LWIP
Compile [100.0%]: main.cpp
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
+------------------------+-------+-------+------+
| Module                 | .text | .data | .bss |
+------------------------+-------+-------+------+
| Fill                   |    51 |     0 |   17 |
| Misc                   | 42814 |  2484 |  141 |
| hal                    |   139 |     0 |    0 |
| platform               |  1066 |     4 |  270 |
| rtos                   |   487 |     4 | 4360 |
| rtos/rtx5              |  7812 |   176 | 1368 |
| targets/TARGET_Realtek |  1753 |     0 | 1968 |
| Subtotals              | 54122 |  2668 | 8124 |
+------------------------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 10792 bytes
Total RAM memory (data + bss + heap + stack): 10792 bytes
Total Flash memory (text + data + misc): 56790 bytes

Image: ./BUILD/REALTEK_RTL8195AM/GCC_ARM/mbed-os-example-blinky.bin
```
Here is my compiler version
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
```
And I didn't change the firmware to the latest one.

closed 2017-08-30 by YiChunHung