ARMmbed/mbed-os #4985: How to print mbed tls debug messages in MBED ?

LMESTM, 2017-08-29:
### Question 

I'd like to see outputs of MBEDTLS macros like MBEDTLS_SSL_DEBUG_MSG.
Has this been plugged into MBED trace module and if so how do I activate it ?

LMESTM, 2017-08-29:
@RonEld @adustm   maybe you guys know how to do it ?

adustm, 2017-08-29:
Hello Laurent, you can see an example here : 
https://github.com/ARMmbed/mbed-os-example-tls/blob/5cca1f74a70855c17cf292f176d0c96db4980df9/tls-client/main.cpp#L48

The debug level is between 0 (nothing) and 4 (every details)

LMESTM, 2017-08-29:
Thanks @adustm . 
So to summarize in this ticket, in order to debug MBEDTLS in mbed we need the program to:
- include mbedtls/debug.h
- define a debug callback (my_debug in the provided example)
- then register this callabck and activate the desired threshold with mbedtls API: 
        mbedtls_ssl_conf_dbg(&_ssl_conf, my_debug, NULL);
        mbedtls_debug_set_threshold(DEBUG_LEVEL);

closed 2017-08-29 by LMESTM

moustafabessar, 2021-02-15:
It doesn't work on matlab mex file
I tried to run the client1.c example on mex file and IT WORKS but debug doesn't work.