ARMmbed/mbed-os #4986: osKernelGetSysTimerFreq returns wrong frequency for nrf51_dk board

maciejbocianski, 2017-08-29:
### Description
- Type: Bug
- Priority: Major

Call of osKernelGetSysTimerFreq returns wrong freq 32678 instead of 16MHz

Function osKernelGetSysTimerFreq in rtos/rtx5/TARGET_CORTEX_M/rtx_system.c
is defined as __WEAK and on this platform is substitiuted by implementation located at
targets/TARGET_NORDIC/TARGET_NRF5/us_ticker.c which return RTC1_CONFIG_FREQUENCY(32768)

---------------------------------------------------------------
## Bug

**Target**
nrf51_dk

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
return value: 16000000 (16MHz)

**Actual behavior**
return value: 32678

**Steps to reproduce**
call function osKernelGetSysTimerFreq 
----------------------------------------------------------------
## Enhancement

**Suggested enhancement**
1. comment out this implementation (this is the only board overloading this function)
2. fix function return value

-----------------------------------------------------------------

## Question

**How to?**

bulislaw, 2017-08-29:
@maciejbocianski can you described how you arrived to this issue
@pan- @nvlsianpu I'm assuming it's because nrf51_dk is not using the SysTick directly for RTX, but 32K is not a lot, most of targets return just the main clock speed in `osKernelGetSysTimerFreq` is the behavior correct for nrf51_dk?

maciejbocianski, 2017-08-29:
I have tried to measure some function call time and expected it will be linearly dependant to CPU speed and the nrf51_dk break this rule significantly.  For other boards (nucleo_f429zi, nucleo_f070rb, k64f) `osKernelGetSysTimerFreq` return values around CPU clock but for nrf51_dk we get value three orders of magnitude lower

pan-, 2017-08-29:
The clock used to generate the tick on the NRF51 is the 32KHz one. I don't see any thing wrong with that. 
However it is a a programming fault to assume that  `osKernelGetSysTimerFreq ` will always return the CPU frequency.

0xc0170, 2017-08-30:
> The clock used to generate the tick on the NRF51 is the 32KHz one. I don't see any thing wrong with that.
However it is a a programming fault to assume that osKernelGetSysTimerFreq will always return the CPU frequency.

I assume this is not a bug, it's correct behaviour in this case. @maciejbocianski please confirm

maciejbocianski, 2017-08-30:
After deeper analysis look ok.  
Not a bug

closed 2017-08-31 by pan-