ARMmbed/mbed-os #4988: Writing a big buffer to a file files

dannybenor, 2017-08-29:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

Writing a buffer to a file
written bytes = 999, expected = 12288
mbed-os-5.5.5 and SD-Driver master head (1d3fa9db3a438f877a46fd166f6b5e8e1b7a85c2) or ( 1d3fa9db3a438f877a46fd166f6b5e8e1b7a85c2 )

dannybenor, 2017-08-29:
@JanneKiiskila

deepikabhavnani, 2017-08-30:
I tried 2K write with simple [application](https://gist.github.com/deepikabhavnani/082554798e60ff1ee040ef468a1f8e27) and it was successful. Please verify if this is failing? If not than please share more details on how to reproduce this and what was ferror for failing case.

geky, 2017-08-30:
Hi @dannybenor, which compiler are you using? Do you think this could be related to https://github.com/ARMmbed/mbed-os/issues/4919?

deepikabhavnani, 2017-08-30:
Some observations after trying #4919 sample
1. Failure is when write is not in multiple of 512 bytes. (Combination of CMD25/CMD24). 
2. Card formatted with format utility / windows does not have this issue. This is seen only if we format card with code `FATFileSystem::format(&bd);`

0xc0170, 2017-08-31:
@dannybenor Any more details for us to reproduce easily? Can you please respond to the comments above?

dannybenor, 2017-09-03:
I am using GCC
All tests can be executed following the instructions in https://github.com/ARMmbed/mbed-client-esfs-example
The tests are in https://github.com/ARMmbed/mbed-client-esfs/tree/master/Test/Unitest
The test failing is writing 12*1024 bytes
After the fails the file system became corrupted.
I did not use FATFileSystem::format(&bd);, only Windows format

dannybenor, 2017-09-04:
Some updates:
I took the last SD-Driver head (https://github.com/ARMmbed/sd-driver/commit/94392432be55e37ffaeb7a8671505470f1f24dee)
The new version cannot load the original sd card "Transcend 8G micro SDHC" where the bug was found.
Old versions can load the card.

The tests pass in a new SD card "Ultra microSDHC UHS-1 16G from SanDisk" with new version but also with the original version where the bug was found (SD driver 1d3fa9db3a438f877a46fd166f6b5e8e1b7a85c2).

I recommend that you get some SD cards that are not from sandisk, like this one
https://www.amazon.com/Transcend-microSDHC-Flash-Memory-TS8GUSDHC6/dp/B001IAMME8

deepikabhavnani, 2017-09-12:
Closing as issue was resolved with https://github.com/ARMmbed/sd-driver/pull/61.
Feel free to open if issue still exists.

closed 2017-09-12 by deepikabhavnani