ARMmbed/mbed-os #4998: usb_host flag in build.py

bcostm, 2017-08-31:
### Description
- Question
- Priority: Blocker

-----------------------------------------------------------------
## Question
The --usb_host flag is no more active in build.py

It has been removed in PR #4313 in commit 2977d0c387ce0390ceca078de1c6ba067fa7e7cd

Is there any plan to add the USB Host support again ?

In fact I would like to clean-up ST USB Host files but I need to build and test my modifications...

0xc0170, 2017-08-31:
cc @theotherjimmy @sg-

theotherjimmy, 2017-08-31:
Could you just add a `--source <path-to-usb-host>`? That should do exactly the same thing as the flag would have.

bcostm, 2017-08-31:
Thanks for the tip. Using --source flags, the build is now OK.

My command for reference is:
`python ./tools/build.py -v -x -j 4 -m NUCLEO_F429ZI -t ARM --rtos --usb --source . --source features\unsupported\USBHost --source features\unsupported\USBHost\USBHostHID --source features\unsupported\USBHost\USBHost --build BUILD`

But now I have an issue with make.py:
` python ./tools/make.py -m NUCLEO_F429ZI -t ARM -n USB_1 --usb`

Note that I have replaced the main.cpp file for USB_1 test by the main.cpp found in tests/usb/host/mouse (previously USB_11 test).

[Error] USBEndpoint.h@22,0:  #5: cannot open source input file "rtos.h": No such file or directory

This points me to the comment in commit 2977d0c387ce0390ceca078de1c6ba067fa7e7cd

> usb host has dependency on rtos1 that is no longer in the
> code base.

So, effectively I think I cannot use USB Host for now.

theotherjimmy, 2017-08-31:
You're welcome.

That does look like a blocking issue. If it really does depend on rtos1, then it should not have compiled without the rtos. Hmmmmmm, strange.

screamerbg, 2017-09-07:
This is the command I use to build it successfully

```
$ mbed compile -t GCC_ARM -m NUCLEO_F429ZI --source . --source features/unsupported/USBHost/ --source features/unsupported/tests/usb/host/mouse/
Building project mbed-os (NUCLEO_F429ZI, GCC_ARM)
Scan: .
Scan: 
Scan: 
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [  0.2%]: Ethernet.cpp
Compile [  0.5%]: AnalogIn.cpp
Compile [  0.7%]: FlashIAP.cpp
Compile [  1.0%]: BusIn.cpp
Compile [  1.2%]: BusOut.cpp
Compile [  1.4%]: BusInOut.cpp
Compile [  1.7%]: I2CSlave.cpp
Compile [  1.9%]: CAN.cpp

[SNIP]

[Warning] USBEndpoint.h@92,13: 'void mbed::Callback<R()>::attach(U*, R (T::*)()) [with T = USBHostSerialPort; U = USBHostSerialPort; R = void]' is deprecated (declared at ./platform/Callback.h:297): Replaced by simple assignment 'Callback cb = func [since mbed-os-5.4] [-Wdeprecated-declarations]
[Warning] USBDeviceConnected.h@109,13: 'void mbed::Callback<R()>::attach(U*, R (T::*)()) [with T = USBHostSerialPort; U = USBHostSerialPort; R = void]' is deprecated (declared at ./platform/Callback.h:297): Replaced by simple assignment 'Callback cb = func [since mbed-os-5.4] [-Wdeprecated-declarations]
Link: mbed-os
Elf2Bin: mbed-os
+--------------------------------+-------+-------+-------+
| Module                         | .text | .data |  .bss |
+--------------------------------+-------+-------+-------+
| USBHost/USBDeviceConnected.o   |   477 |     0 |     0 |
| USBHost/USBEndpoint.o          |   631 |     0 |     0 |
| USBHost/USBHost.o              |  5426 |     0 |     4 |
| USBHostHID/USBHostMouse.o      |  1025 |     0 |     0 |
| USBHostHub/USBHostHub.o        |  1218 |     0 |     0 |
| USBHostSerial/USBHostSerial.o  |    16 |     0 |     0 |
| [fill]                         |   119 |     0 |    15 |
| [lib]/libc.a                   | 27554 |  2204 |    56 |
| [lib]/libgcc.a                 |  3728 |     0 |     0 |
| [lib]/libm.a                   |    88 |     0 |     0 |
| [lib]/misc                     |   236 |    16 |    28 |
| drivers/I2C.o                  |    16 |     0 |     0 |
| drivers/SPI.o                  |   102 |     4 |   100 |
| features/FEATURE_LWIP          |    44 |     0 | 12556 |
| features/netsocket             |    22 |     0 |     0 |
| hal/mbed_gpio.o                |    58 |     0 |     0 |
| hal/mbed_pinmap_common.o       |   248 |     0 |     0 |
| hal/mbed_ticker_api.o          |   214 |     0 |     0 |
| hal/mbed_us_ticker_api.o       |    44 |     0 |    24 |
| main.o                         |   299 |     4 |    28 |
| platform/mbed_alloc_wrappers.o |    16 |     0 |     0 |
| platform/mbed_assert.o         |    85 |     0 |     0 |
| platform/mbed_board.o          |   192 |     0 |     0 |
| platform/mbed_critical.o       |   243 |     0 |     5 |
| platform/mbed_error.o          |    44 |     0 |     1 |
| platform/mbed_retarget.o       |   906 |     4 |   264 |
| platform/mbed_wait_api_rtos.o  |    86 |     0 |     0 |
| rtos/Mutex.o                   |   174 |     0 |     0 |
| rtos/Semaphore.o               |   153 |     0 |     0 |
| rtos/TARGET_CORTEX             | 10787 |   176 |  5728 |
| rtos/Thread.o                  |   798 |     0 |     0 |
| rtos/rtos_idle.o               |    20 |     4 |     0 |
| targets/TARGET_STM             | 14537 |     4 |  2195 |
| Subtotals                      | 69606 |  2416 | 21004 |
+--------------------------------+-------+-------+-------+
Total Static RAM memory (data + bss): 23420 bytes
Total Flash memory (text + data): 72022 bytes

Image: ./BUILD/NUCLEO_F429ZI/GCC_ARM/mbed-os.bin
```

bcostm, 2017-09-07:
Thanks! It work for me too using this command.

closed 2017-09-07 by bcostm