ARMmbed/mbed-os #5002: UART overflow cause system hang?

shayo, 2017-08-31:
- Type: Question

How do I set the size of the UART Serial buffer?
I encountered an issue where directly after a soft boot, the device (NRF52832) is trying to send a bunch of stuff over TX and then it hangs at some point.

I didn't see an API in Serial object to change that.
Where is it defined in the low level implementation?

0xc0170, 2017-09-01:
How are you using the serial ? Can you share the small code snippet even pseudocode? Are you using UARTSerial class?

The nordic implementation https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/serial_api.c#L150, you can see there UART Handler.

shayo, 2017-09-01:
I am using the Serial class.

When I define 
Serial pc(TX,RX)

Is the default async ? if so, what is the size of the buffer that is allocated?

0xc0170, 2017-09-01:
Serial class is not buffered (sw buffer). Hw buffered might be for some targets (fifo enabled).

Look at `UARTSerial`

> I encountered an issue where directly after a soft boot, the device (NRF52832) is trying to send a bunch of stuff over TX and then it hangs at some point.

To reproduce this , please share a code snippet. Where it hangs?

shayo, 2017-09-01:
OK. I was wrong. This is not related to the Serial class.
I had a wait statement after the pc.printf, and it seems like the "wait" statement is the culprit.
Is there a reason why wait will fail after a soft reboot?

jrobeson, 2017-11-26:
@shayo: Can you still reproduce this in the latest 5.6?

shayo, 2017-11-26:
I'm am out until next week. I will test this once I'm back.

shayo, 2017-12-06:
Initial tests I ran today indicate this is likely to been solved.
I'm going to close this for now.

closed 2017-12-06 by shayo