ARMmbed/mbed-os #5004: Ticker test fails when run under Greentea on NUCLEO_F070RB

maciejbocianski, 2017-09-01:
Below code fails when run under greentea environment
When comiled and run manually (mbed compile -vv -t ARM -m nucleo_f070rb) without greentea works fine

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
only NUCLEO_F070RB

**Toolchain:**
all

**Expected behavior**
Both counters are equal to 16
**Actual behavior**
Only first counter is equal to 16 second one (multi_counter2) has random value below 16
**Steps to reproduce**
paste below code into ticker test and run below code
 mbed test -t ARM -m nucleo_f070rb -n tests-mbed_drivers-ticker -vv

-----------------------------------------------------------------

## Question

**Any ideas why below test run under greentea fails?**

Fails in line:
    // (2) TEST_ASSERT_EQUAL(TICKER_COUNT, multi_counter2);

```
#include "mbed.h"

#define TICKER_COUNT 16
volatile uint32_t multi_counter1;
volatile uint32_t multi_counter2;

void wait_callback_multi1(void)
{
    multi_counter1++;
}

void wait_callback_multi2(void)
{
    multi_counter2++;
}

/** Test many tickers run simultaneously

    Given an 16 tickers
    When schedule them
    Then tickers properly execute callback
 */
void test_multi_ticker(void)
{
    Ticker ticker[TICKER_COUNT];

    multi_counter1 = 0;
    for(int i = 0; i < TICKER_COUNT; i++) {
        ticker[i].attach_us(callback(wait_callback_multi1), 100 * 1000 + i * 15);
    }

    Thread::wait(110);
    for(int i = 0; i < TICKER_COUNT; i++) {
            ticker[i].detach();
    }

    // (1) TEST_ASSERT_EQUAL(TICKER_COUNT, multi_counter1);
    printf("(1) expected: %d was: %d\r\n", TICKER_COUNT, multi_counter1);

    multi_counter2 = 0;
    for(int i = 0; i < TICKER_COUNT; i++) {
        ticker[i].attach_us(callback(wait_callback_multi2), 100 * 1000);
    }

    Thread::wait(110);
    for(int i = 0; i < TICKER_COUNT; i++) {
            ticker[i].detach();
    }

    // (2) TEST_ASSERT_EQUAL(TICKER_COUNT, multi_counter2);
    printf("(2) expected: %d was: %d\r\n", TICKER_COUNT, multi_counter2);
}

int main() {
    test_multi_ticker();
}
```

0xc0170, 2017-09-01:
cc @bcostm  @adustm  @LMESTM @jeromecoutant

LMESTM, 2017-09-01:
@0xc0170  Considering it works fine when compiled and run manually, could explain explain the difference of running an app or test from greentea ?

0xc0170, 2017-09-01:
> @0xc0170 Considering it works fine when compiled and run manually, could explain explain the difference of running an app or test from greentea ?

This: `-DMBED_MEM_TRACING_ENABLED=1 -DMBED_HEAP_STATS_ENABLED=1`. you should be able to reproduce it.

maciejbocianski, 2017-09-01:
I'm not sure about this difference. 
I put below code to tested code and get compile errors.
So this flags are not defined under greentea path

```
#ifndef MBED_MEM_TRACING_ENABLED
#error "!!!!!!!!! MBED_MEM_TRACING_ENABLED not defined"
#endif

#ifndef MBED_MEM_TRACING_ENABLED
#error "!!!!!!!!! MBED_HEAP_STATS_ENABLED not defined"
#endif
```

0xc0170, 2017-09-01:
cc @studavekar

LMESTM, 2017-09-01:
@maciejbocianski  one more point : what is the version used - are you running from mbed/master ?

maciejbocianski, 2017-09-01:
@LMESTM yes from master

LMESTM, 2017-09-04:
@maciejbocianski @0xc0170  I reproduce the issue on master. But what I can see is that the issue happens whether greentea is used or not. I debugged and could not find out anything wrong in targets/TARGET_STM/us_ticker_16b.c so I ended up reverting #4644 and the test is OK. I think #4644 included some simplification of our code that should not have been done. I re-introduced the original implementation of us_ticker_set_interrupt and the test is OK now.

LMESTM, 2017-09-12:
@0xc0170 @maciejbocianski  #5012 is the fix for the issue. and has been merged, so I propose to close this issue@0xc0170 @maciejbocianski  #5012 is the fix for the issue. and has been merged, so I propose to close this issue

maciejbocianski, 2017-09-12:
issue fixed in https://github.com/ARMmbed/mbed-os/pull/5012

closed 2017-09-12 by maciejbocianski