ARMmbed/mbed-os #5005: EFM32GG: Simple blinky app doesn't work

bulislaw, 2017-09-01:
### Description
- Type: Bug 
- Priority: Major

## Bug

**Target**
EFM32GG

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.x

**mbed-os sha:**
`4978a08ae274`

** Code **

```
#include "mbed.h"

DigitalOut led1(LED1);

// main() runs in its own thread in the OS
int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
}
```

**Expected behavior**
LED is blinking for all the build profiles.

**Actual behavior**
LED stays on for develop and release and blinks for debug.

**Details**
The issue is around sleep, for develop and release profiles the platforms calls sleep inside the idle handler: https://github.com/ARMmbed/mbed-os/blob/master/rtos/rtos_idle.c#L26 for debug it's disabled.

The EFM32 hal_sleep call looks like that:
```
void hal_sleep(void)
{
    if (sleep_block_counter[0] > 0) {
        /* Blocked everything below EM0, so just return */
        return;
    } else if (sleep_block_counter[1] > 0) {
        /* Blocked everything below EM1, enter EM1 */
        EMU_EnterEM1();
    } else if (sleep_block_counter[2] > 0) {
        /* Blocked everything below EM2, enter EM2 */
        EMU_EnterEM2(true);
    } else {
        /* Blocked everything below EM3, enter EM3 */
        EMU_EnterEM3(true);
    } /* Never enter EM4, as mbed has no way of configuring EM4 wakeup */
    return;
}
```
it seems that the board goes into a too deep sleep and cannot wake up on the systick. When I modify the sleep function to only call `EMU_EnterEM1();` everything works fine.

CC: @0xc0170 (could you tag SiLab people please)

bulislaw, 2017-09-04:
@0xc0170 ping

0xc0170, 2017-09-04:
cc @stevew817 @akselsm

stevew817, 2017-09-04:
@bulislaw I think this would be best solved by 'blocking' everything under EM1 out of the box when using RTOS. Else you will, indeed, switch off the clock to systick.

How far along are mbed OS' efforts on tickless?

bulislaw, 2017-09-05:
We are almost done, it solves the problem of sleeping during idle as we sets up a different wake up source https://github.com/ARMmbed/mbed-os/pull/4991 but we're not sure about enabling strategy yet.

stevew817, 2018-01-15:
@bulislaw Can we close this now that #5581 was merged?

closed 2018-01-15 by 0xc0170