ARMmbed/mbed-os #5007: Single byte write to I2C EEPROM failing on NCS36510

maclobdell, 2017-09-01:
### Description
- Type: Bug
- Related issue: n/a
- Priority: Major

---------------------------------------------------------------
## Bug
Fails [CI Test Sheild](https://github.com/ARMmbed/ci-test-shield) I2C test of Single Byte writes to EEPROM

NCS36510-GCC_ARM | NCS36510      | tests-api-i2c | I2C -  EEProm WR Single Byte       | 0      | 2      | FAIL   | 0.2                |

**Target**
NCS36510

**Toolchain:**
n/a - tested GCC, but expect to be same on all toolchain

**mbed-os sha:**
5.5.6

**Expected behavior**
  This test case checks that the I2C can write and read a single byte to the EEPROM on the ci test shield.  It uses different hal apis than the other tests which write more amounts of data.  It should pass.  It passes on other platforms tested.  

**Actual behavior**

In the test case, the single_byte_WR function tries to write a single byte of data using the EEPROM library's write(address,test); method, which then calls m_i2c.write(m_i2cAddress, values, 3), which calls  i2c_write(&_i2c, address, data, length, stop).  At the conclusion of the i2c_write function, the EEPROM acknowledges the write with a ACK.  

Next the EEPROM library uses its waitForWrite() routine which initiates a write cycle on the bus by  sending the address of the EEPROM.  The EEPROM then responds with a NACK indicating it is busy.  Every 1ms another request is sent, until the EEPROM is ready and responds with an ACK.  

For some reason on this platform, the first time the the request is sent to check if the EEPROM is busy, the I2C on NCS36510 seems to think it received a ACK, but on the bus the signals show that it was sent a NACK.  So, it tries to go ahead with the next set of cycles without the EEPROM being ready.

**Steps to reproduce**
mbed import ci-test-shield
mbed test -m NCS36510 -t GCC_ARM -n tests-api-i2c* --app-config .\mbed_app.json

**Possible fix**
Adding 100us wait in int i2c_write(i2c_t *obj, int address, const char *data, int length, int stop) of mbed-os\targets\TARGET_ONSEMI\TARGET_NCS36510\i2c_api.c, fixes this issue.  

I suspect that a more elegant and robust solution is needed.  It seems that there is some time between when the function completes and when the status register is valid.  
I tried  wait until the command overflow ends, but that didn't work.  
 
The root cause of the problem is still unknown and more tests need to be done.

```
int i2c_write(i2c_t *obj, int address, const char *data, int length, int stop)
{
    int Count, status;
    const char WriteData = (address & I2C_READ_WRITE_BIT_MASK) & 0xFF;

    /* Send start bit */
    status = fI2cStart(obj);
    if(status) {
        /* Error sending start bit */
        return status;
    }

    /* Send address | write */
    Count = fI2cWriteB(obj, &WriteData, 1);
    if(Count != 1) {
        /* Error sending address */
        return Count;
    }

    /* Sens command, [data] */
    Count = fI2cWriteB(obj, data, length);
    if(Count != length) {
        /* Error sending address */
        return Count;
    }

    if(stop) {    /* If stop requested */
        /* Send stop bit */
        status = fI2cStop(obj);
        if(status) {
            /* Error sending stop bit */
            return status;
        }
    }

    wait_us(100);  //adding this allows time for the status bits to get set properly??  
    
    return Count;
}
```

**Possible Fix #2**

Alternatively, the issue might actually be with the i2c_byte_write command which should wait for the status register to be ready to read before checking it.  

```
int i2c_byte_write(i2c_t *obj, int data)
{
    int Count;
    Count = fI2cWriteB(obj, (const char *)&data, 1);
    if(Count != 1) {
        return Count;
    }

    while(obj->membase->STATUS.WORD & I2C_STATUS_CMD_FIFO_OFL_BIT); /* Wait till command overflow ends */
       //is the above not waiting for the status bits to be updated?
       wait_us(100);  //adding this allows time for the status bits to get set properly??  
    
    if(obj->membase->STATUS.WORD & I2C_STATUS_BUS_ERR_BIT) {
        /* Bus error means NAK received */
        return 0;
    } else {
        /* ACK received */
        return 1;
    }
}
```

danclement, 2017-09-01:
@maclobdell Thank you very much for the very detailed bug report.  We will look into it.

jacobjohnson-ON, 2017-10-04:
I ran some tests on this, and when outside of the test suite, the single byte write is working fine.  My measurements even suggest that it is working fine inside the test suite, but it isn't getting recorded properly. 

I have attached a PDF with the results of testing the single byte write outside of the test suite.  I have also attached the captured data from a protocol analyzer which shows the correct bytes being read out from the EEPROM memory for the single byte write. 

[I2C Debugging.pdf](https://github.com/ARMmbed/mbed-os/files/1356565/I2C.Debugging.pdf)

[protocol_decode.xlsx](https://github.com/ARMmbed/mbed-os/files/1356563/protocol_decode.xlsx)

maclobdell, 2017-10-09:
Thanks a lot Jake.  This is good news.  It is unusual that the problem is so repeatable with the ci-test-shield tests.  Looking at the protocol traffic during the failing test, I do see the error.  Could there be a board-to-board variation or maybe there is a system timing difference.

danclement, 2017-10-27:
There were several issues/bugs in the I2C driver.  I've been working on fixes and am now going through testing and will submit a pull request soon if all tests pass.

danclement, 2017-11-06:
@maclobdell can you close this issue now that #5394 is merged?

closed 2017-11-07 by 0xc0170