ARMmbed/mbed-os #5009: Question: turning off BLE ?

shayo, 2017-09-02:
Question: How can I turn off BLE completely after initialization?
I'm using a custom board running with NRF52832

I use the following to initialize the ble stack:
BLE& ble = BLE::Instance(BLE::DEFAULT_INSTANCE);
ble.init(bleInitComplete);

Simply stopping advertising doesn't work (i.e.,  if the central already knows the address, it can still connect).
How do I fully shutdown any transmission?

0xc0170, 2017-09-04:
cc @pan-

shayo, 2017-09-04:
I ended up using ble.shutdown(), to clean up the stack.
I am not sure if the softdevice will turn off radio completely if there are no advertising/services declared. Can someone confirm?

pan-, 2017-09-12:
> Simply stopping advertising doesn't work (i.e., if the central already knows the address, it can still connect).

If advertising are stopped a central shall not be able to connect to the device. There might be an issue with the function disabling advertising. 

Can you post some sample code ?

shayo, 2017-09-12:
I believe I did something like:

  BLE::Instance(BLE::DEFAULT_INSTANCE).gap().stopAdvertising();

shayo, 2017-09-21:
Any update on this?
Can you please confirm that if BLE is not advertising, all radio components are off?

pan-, 2017-10-11:
Calling `Gap::stopAdvertising` should be sufficient to stop the advertising.

> if the central already knows the address, it can still connect

That's not how the connection process work in BLE. A peripheral listen for connection requests after any connectable advertisement, similarly a central send a connection request right after the reception of an advertising packet it want to connect to. A central does not send random connection requests. 

It is important to precise that the connection window is quite small too (interframe spacing between the advertisement packet and the connection request is 150us).

0xc0170, 2018-06-26:
> Calling Gap::stopAdvertising should be sufficient to stop the advertising.

We believe this issue has been resolved.

closed 2018-06-26 by 0xc0170