ARMmbed/mbed-os #5010: Support for FRDM-KL28Z (?)

larsgk, 2017-09-02:
### Description
- Type: Enhancement | Question
- Priority: Blocker

---------------------------------------------------------------
## Enhancement

Is there already support for FRDM-KL28Z or what is needed to add the target and what can be done to get it ASAP?

I bought a bunch of boards I would have liked to get running with software from https://github.com/empirikit/empirikit-controller before next week - not realistic, I know ;)

0xc0170, 2017-09-04:
KL27Z is supported (using MCUXpresso drivers). I would take that one as a template and port KL28Z.

cc @mmahadevan108

mmahadevan108, 2017-09-05:
@larsgk. We do not have plans to add mbed supporf for KL28Z till later this year. 
As Martin recommended you could use the KL27Z port as reference and use the KL28Z SDK release available on https://mcuxpresso.nxp.com/en/welcome to add support.
I will be happy to work with you (answer questions, review etc.) on this. Thanks.

larsgk, 2017-09-26:
Hi @mmahadevan108 and @0xc0170 and thanks :)

I missed the KL28Z opportunity for the conference I wanted to demo at - but I am going to try to do some integration within a week or so.  Would questions be best sent here or somewhere else (e.g. in some slack channel)?

mmahadevan108, 2017-09-28:
Feel free to post questions to the below section:
https://os.mbed.com/teams/NXP/community/

omarisai, 2017-12-27:
Hi @0xc0170 , did you ported the code for KL28Z?

0xc0170, 2018-01-08:
> Hi @0xc0170 , did you ported the code for KL28Z?

No, KL28Z is still not supported as indicated above

larsgk, 2018-01-09:
Is anyone working on it (or have it assigned)?

omarisai, 2018-04-05:
I started to work on this, but I don't have to much time. Right now the rtos is working and also uart, when I have some time I will continue with I2C and SPI.

0xc0170, 2018-04-05:
> I started to work on this, but I don't have to much time. Right now the rtos is working and also uart, when I have some time I will continue with I2C and SPI.

It might help others if you share your branch (if there is anyone who is willing to help they can contribute via sending a pull request). Thanks for sharing the info!

omarisai, 2018-04-05:
this is the branch [mbed-os](https://github.com/omarisai/mbed-os)

mmahadevan108, 2018-08-09:
Can we close this ticket. Please submit any PR's you may have that adds support for KL28Z via a separate ticket.

maclobdell, 2018-08-16:
Closing this ticket because it seems the initial question has been answered.  Please re-open if needed.

closed 2018-08-16 by maclobdell