ARMmbed/mbed-os #5014: Cannot compile for nRF51-DK anymore w/ GCC_ARM

janjongboom, 2017-09-04:
Take an application, f.e. https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-ble-HeartRate/ using mbed OS 5.5.5

Compile with GCC_ARM for nRF51-DK:

```
Compile [ 57.0%]: rtx_evflags.c
[Error] cc0lgVyk.s@677,0: cannot honor width suffix -- `mov r1,#0'
[Error] cc0lgVyk.s@679,0: cannot honor width suffix -- `mov r0,#0'
[Error] cc0lgVyk.s@727,0: cannot honor width suffix -- `mov r1,#4'
[Error] cc0lgVyk.s@729,0: cannot honor width suffix -- `neg r1,r1'
[Error] cc0lgVyk.s@731,0: cannot honor width suffix -- `mov r5,#4'
[Error] cc0lgVyk.s@737,0: cannot honor width suffix -- `mov r1,#3'
[Error] cc0lgVyk.s@739,0: cannot honor width suffix -- `neg r1,r1'
[Error] cc0lgVyk.s@741,0: cannot honor width suffix -- `mov r5,#3'
[Error] cc0lgVyk.s@743,0: cannot honor width suffix -- `neg r5,r5'
[Error] cc0lgVyk.s@754,0: cannot honor width suffix -- `orr r5,r2'
```

0xc0170, 2017-09-04:
What is the version for GCC ? I remember seeing this errors when using older GCC (in my case it was 4.x).

janjongboom, 2017-09-04:
@0xc0170 Good point. Works with GCC6.

closed 2017-09-04 by janjongboom