ARMmbed/mbed-os #5015: NRF51_DK_BOOT target is broken

janjongboom, 2017-09-04:
1. Take https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-ble-HeartRate/
2. Compile with ARMCC for NRF51_DK => works fine
3. Compile with ARMCC for NRF51_DK_BOOT => does not work. Binary is 1/3 of the non-boot image and board does not start.

Tested on mbed OS 5.5.5.

I think this has been broken since mbed OS 5, as even with mbed OS 5.2.3 it does not produce a working binary.

I tested the mbed OS 2 version of this program ([here](https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_HeartRate/)) and that one does not compile (against rev126 and rev150):

```
Error: Cannot open source input file "device.h": No such file or directory in "extras/mbed_abea610beb85/platform.h", Line: 19, Col: 21
```

0xc0170, 2017-09-04:
cc @pan- @nvlsianpu @anangl

janjongboom, 2017-09-04:
FYI, this is what gets linked in on NRF51_DK:

```
+-----------------------+-------+-------+------+
| Module                | .text | .data | .bss |
+-----------------------+-------+-------+------+
| Fill                  |   165 |    14 |   53 |
| Misc                  | 35916 |  2489 |  218 |
| drivers               |   832 |     0 |    0 |
| events                |    54 |     0 |    0 |
| events/equeue         |  1372 |     0 |   93 |
| features/FEATURE_BLE  | 16664 |     5 |  572 |
| hal                   |   686 |     0 |   48 |
| platform              |  1633 |     4 |  265 |
| rtos                  |   463 |     4 | 4360 |
| rtos/rtx5             |  9102 |   176 | 1368 |
| targets/TARGET_NORDIC | 23858 |   140 | 2187 |
| Subtotals             | 90745 |  2832 | 9164 |
+-----------------------+-------+-------+------+
```

and this is what gets linked in on NRF51_DK_BOOT:

```
+-----------------------+-------+-------+------+
| Module                | .text | .data | .bss |
+-----------------------+-------+-------+------+
| Fill                  |   118 |    10 |   45 |
| Misc                  | 35888 |  2489 |  241 |
| drivers               |   814 |     0 |    0 |
| events                |    72 |     0 |    0 |
| events/equeue         |  1350 |     0 |   93 |
| features/FEATURE_BLE  | 24256 |    13 | 1200 |
| hal                   |   634 |     0 |   24 |
| platform              |  1540 |     4 |  265 |
| rtos                  |   463 |     4 | 4360 |
| rtos/rtx5             |  9186 |   176 | 1368 |
| targets/TARGET_NORDIC |  2360 |    12 |  200 |
| Subtotals             | 76681 |  2708 | 7796 |
+-----------------------+-------+-------+------+
```

Seems to be missing a bunch of stuff in TARGET_NORDIC.

NRF51_DK_OTA fails to build on mbed OS 5.5.5 with GCC6: `[ERROR] Starting addresses are different`

SenRamakri, 2018-12-04:
@janjongboom - Is this still an issue?

janjongboom, 2018-12-19:
@SenRamakri I don't have a nRF51 with me, but I'd assume so.

deepikabhavnani, 2018-12-28:
NRF51_DK_BOOT still fails to build on master (after 5.11 release)

```
C:\Users\deebha02\work_mbed\nordic_icsr_fix>mbed compile -m NRF51_DK_OTA -t ARM
Compile [  5.6%]: BLE.cpp
[Error] DFUService.h@26,0:  #5: cannot open source input file "dfu_app_handler.h": No such file or directory
[ERROR] "./mbed-os/features/FEATURE_BLE/ble/services/DFUService.h", line 26: Error:  #5: cannot open source input file "dfu_app_handler.h": No such file or directory
.\mbed-os\features\FEATURE_BLE\source\BLE.cpp: 0 warnings, 1 error

[mbed] ERROR: "c:\yotta\python\python.exe" returned error.
       Code: 1
       Path: "C:\Users\deebha02\work_mbed\nordic_icsr_fix"
       Command: "c:\yotta\python\python.exe -u C:\Users\deebha02\work_mbed\nordic_icsr_fix\mbed-os\tools\make.py -t ARM -m NRF51_DK_OTA --source . --build .\BUILD\NRF51_DK_OTA\ARM --stats-depth 2"
       Tip: You could retry the last command with "-v" flag for verbose output
```

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .

jrobeson, 2021-06-17:
this bug should be closed altogether. There's no such target anymore. The FOTA stuff with mcuboot in the experimental ble services repository is the way forward.

closed 2021-06-18 by 0xc0170