ARMmbed/mbed-os #5017: Callback: comparision fails

0xc0170, 2017-09-04:
### Description
- Type: Bug 
- Priority: Major 

How to reproduce:

```
#include "mbed.h"
#include "rtos.h"

void donothing()
{

}

int main()
{
    Callback<void()> func1;
    Callback<void()> func2;
    func1 = donothing;
    func2 = donothing;
    if (func1 != func2) {
        error("Should be equal");
    }
}
```

I had similar code in one patch that started failing. I was looking for API in Callback that would check if Callback was assigned the same function (the default one), the only way there I found was to use comparision operators. Those  however fail.

What I noticed that when those two objects are compared, `_obj` is always different. Seems to me that `generate()` method does not generate properly callback - does not pollute all members?

@geky @pan-

0xc0170, 2017-09-05:
Also operator== seems to be broken, Callback is non-POD and we use memcmp.

geky, 2017-09-05:
Thanks for adopting the workaround, here's a quick fix for the Callback class: https://github.com/ARMmbed/mbed-os/pull/5027

PR #5027 (2017-09-27): cb: Fix uninitialized memory used for equality check
Fixed by zeroing the uninitialized memory. A more "correct" approach may be to add an additional op for equality, and use the "==" operator on the F type in the generate function. But this adds the requirement that F supports equality.

cc @0xc0170, @pan- 
fixes #5017
closed 2017-09-27 by theotherjimmy