ARMmbed/mbed-os #5023: Cannot reconnect a LoWPANNDInterface

Mathgoat, 2017-09-05:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
K64F

**Hardware**
Firefly 6LoWPAN shield +K64F

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.1.1

**Expected behavior**
 As the EthernetInterface, the LoWPANNDInterface should be able to reconnect after a deconnection.

**Actual behavior**
The LoWPANNDInterface can't be reconnected after a deconnection.

**Steps to reproduce**
```
#include "mbed.h"
#include "NanostackInterface.h"
#include "NanostackRfPhyAtmel.h"

int main() {
 	LoWPANNDInterface interface;
	NanostackRfPhyAtmel rf_phy(ATMEL_SPI_MOSI, ATMEL_SPI_MISO,
	ATMEL_SPI_SCLK,
 	ATMEL_SPI_CS,
 	ATMEL_SPI_RST, ATMEL_SPI_SLP, ATMEL_SPI_IRQ, ATMEL_I2C_SDA,
 	ATMEL_I2C_SCL);
 	interface.initialize(&rf_phy);
 	interface.connect();
 	interface.disconnect();
 	rf_phy.rf_unregister(); // "Multiple registrations of NanostackRfPhyAtmel not supported" if not done.
 	interface.connect(); // stuck into connect_semaphore.wait(osWaitForever);
 }
```

Without the "rf_phy.rf_unregister" line, I get the error : "Multiple registrations of NanostackRfPhyAtmel not supported"

With the "rf_phy.rf_unregister()" line, it looks like the "connect_semaphore.wait(osWaitForever)" in LoWPANNDInterface::connect() method is stuck.

Is there something I do wrong ?

0xc0170, 2017-09-05:
cc @SeppoTakalo

mikaleppanen, 2018-08-29:
Corrected in: https://github.com/ARMmbed/mbed-os/pull/7878 and https://github.com/ARMmbed/mbed-os/pull/7818

closed 2018-10-02 by adbridge