ARMmbed/mbed-os #5024: Odin fails networking tests with heap and stack tracking

sarahmarshy, 2017-09-05:
### Description
- Type: Bug

I am using my fork of mbed OS that adds networking tests for Odin - https://github.com/ARMmbed/mbed-os/pull/4795. Here are the commands run:
```
mbed test --compile -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -m auto -t IAR -n tests-netsocket-tcp_hello_world,tests-netsocket-socket_sigio -v
mbed test -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -m auto -t IAR -n tests-netsocket-tcp_hello_world,tests-netsocket-socket_sigio -v --run
```

These commands pass when the heap and stack stats macros are not added. They fail when the macros are used, which is how the commands are run in CI. 

---------------------------------------------------------------
## Bug

**Target**
Odin W2

**Toolchain:**
IAR

**Expected behavior**
The tests pass.

**Actual behavior**
They fail with a timeout.

**Steps to reproduce**
Clone my fork of mbed OS and check out the `test-configs` branch.  Run the following commands: 
```
mbed test --compile -m auto -t IAR -n tests-netsocket-tcp_hello_world,tests-netsocket-socket_sigio -v
mbed test -m auto -t IAR -n tests-netsocket-tcp_hello_world,tests-netsocket-socket_sigio -v --run
```

These should pass.

Then run:
```
mbed test --compile -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -m auto -t IAR -n tests-netsocket-tcp_hello_world,tests-netsocket-socket_sigio -v
mbed test -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -m auto -t IAR -n tests-netsocket-tcp_hello_world,tests-netsocket-socket_sigio -v --run
```
These will fail.

0xc0170, 2017-09-05:
cc @andreaslarssonublox @andreaspeterssonublox

studavekar, 2017-11-03:
@sg- this will block testing on Odin.

sg-, 2017-11-03:
Let's get this fixed then 👍

andreaslarssonublox, 2017-11-08:
@sarahmarshy, is this a problem with 'mbed test' not accepting macros or something with the ODIN target?

sarahmarshy, 2017-11-08:
It is a problem with Odin. The tests accept macros, but something about using them makes the tests fail on Odin.

andreaslarssonublox, 2017-11-29:
I tested this with a NUCLEO_F429ZI board and got the same error as with ODIN:

	C:\_issues\5024\mbed-os>mbed test -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -m auto -t IAR -n tests-mbed_drivers-ticker -vv --run
	[mbed] Working path "C:\_issues\5024\mbed-os" (program)
	[mbed] WARNING: Could not find mbed program in current path "C:\_issues\5024\mbed-os".
	[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
	---
	[mbed] Detected "NUCLEO_F429ZI" connected to "D:" and using com port "COM19"
	[mbed] Exec "mbedgt --test-spec C:\_issues\5024\mbed-os\BUILD\tests\NUCLEO_F429ZI\IAR\test_spec.json -n tests-mbed_drivers-ticker -V -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1" in C:\_issues\5024\mbed-os
	Usage: mbedgt [options]

	mbedgt: error: no such option: -D
	[mbed] ERROR: "mbedgt" returned error code 2.
	[mbed] ERROR: Command "mbedgt --test-spec C:\_issues\5024\mbed-os\BUILD\tests\NUCLEO_F429ZI\IAR\test_spec.json -n tests-mbed_drivers-ticker -V -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1" in "C:\_issues\5024\mbed-os"
	---

0xc0170, 2017-11-29:
@andreaslarssonublox not sure I am reading the snippet correctly, but its regarding `-D` option error. Please follow the commands that are shared above by @sarahmarshy in her first post. It first builds and then run (test --compile first command, second one is test --run)

screamerbg, 2018-09-10:
@sarahmarshy Is this still an issue with the latest u-blox ODIN driver?

asifrizwanubx, 2018-09-14:
Internal Ref: TE_OM2-127 (in wait for info state)

AmmadRehmat, 2019-04-05:
@screamerbg @0xc0170  is this issue still there? Can we close it if its not reproduceable anymore?

0xc0170, 2019-04-05:
Too old to be still active, will need to be reopened via new ticket with latest test results. I'll close this one.

0xc0170, 2019-04-05:
Just in case, cc @SeppoTakalo

closed 2019-04-05 by 0xc0170