ARMmbed/mbed-os #5034: Remove NonCopyable from SPI / I2C / etc.

janjongboom, 2017-09-06:
### Description

- Type: Enhancement
- Related issue: #4594
- Priority: Major

Since mbed OS 5.5.2 we've marked SPI and I2C as NonCopyable. I understand the reasoning behind this, but it has broken many mbed libraries, which cannot be easily updated. This is user hostile and bad for the platform.

Some examples that are broken now and that I found with a quick Google search:

* HelloMQTT
* AT45 SPI Flash driver
* https://developer.mbed.org/questions/78698/Program-for-STM32F411RE-wont-compile/
* https://developer.mbed.org/forum/helloworld/topic/27921/?page=1#comment-53030
* https://developer.mbed.org/questions/78589/Compiling-error-with-IBMIoTClientLibrary/

Some of our partners / other devs were apparently not aware, and they have only now started patching:

* https://developer.mbed.org/teams/MaximIntegrated/code/MAX14690/graph
* https://developer.mbed.org/users/kenjiArai/code/Si5351A/rev/8c63d15c8c2e

There are probably hundreds of libraries that are broken now (@andcor02 had one too), the error messages are cryptic and this is very bad for the mbed platform. I'm fine throwing a deprecation warning, but breaking these libraries in a *minor* release is bad.

@sg-

0xc0170, 2017-09-06:
cc @pan- @bulislaw

pan-, 2017-09-06:
> Since mbed OS 5.5.2 we've marked SPI and I2C as NonCopyable. I understand the reasoning behind this, but it has broken many mbed libraries, which cannot be easily updated. This is user hostile and bad for the platform.

No this is undefined behavior hostile, if these library doesn't compile now then it means they were executing code with undefined behavior. 

> I'm fine throwing a deprecation warning, but breaking these libraries in a minor release is bad.

The important part to understand is that the code was **already** broken even if it wasn't visible, a change of compiler version, optimization or just subclassing and use of the SPI, I2C, ... classes might have been enough to exhibit the undefined behavior. 

However I do agree that the impact on user libraries was minimized and the communication could have been better,

sg-, 2017-09-06:
While a great addition, this never should have been part of a patch release. @pan- can you and @janjongboom work to get a blog created about this that goes out before Mbed OS 5.6? Let's also make sure a API page gets created in the references for the 5.6 release.

@0xc0170 @theotherjimmy

janjongboom, 2017-11-11:
And today also learned that Zigbee library is broken due to this. Can we retract this patch, it's breaking libraries left and right and gives a bad view of our ecosystem.

0xc0170, 2017-11-30:
Fixed via https://github.com/ARMmbed/mbed-os/pull/5485

0xc0170, 2017-11-30:
>  @pan- can you and @janjongboom work to get a blog created about this that goes out before Mbed OS 5.6? Let's also make sure a API page gets created in the references for the 5.6 release.

One more item left, @janjongboom @pan- Please coordinate for the page in the docs and blog

closed 2018-08-27 by deepikabhavnani