ARMmbed/mbed-os #5041: Error with SPI frequency set by KL25Z implementation

maclobdell, 2017-09-06:
### Description
- Type: Bug 
- Related issue: n/a
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
KL25z

**Expected behavior**

spi_frequency() should set the SPI clock frequency to be very close to the desired frequency per the argument.

**Actual behavior**
If you pass 8MHz as your selected frequency, prescaler scales it down to 6 MHz. And if you send 16 MHz, it gets scaled down to exactly 8 MHz. Reason being half bus-clock frequency doesn't seem to be correct. 

Suspect there is an error in the calculation of the prescaler here https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_KLXX/TARGET_KL25Z/spi_api.c#L86

Link to Reference Manual for background information:  http://cache.freescale.com/files/32bit/doc/ref_manual/KL25P80M48SF0RM.pdf , check page 665, section 'SPI0_BR field descriptions'. 

**Steps to reproduce**
Something like this should show the problem.

SPI spi(D11, D12, D13); // mosi, miso, sclk

spi.frequency(8000000);

spi.write(0x8F);

Check the spi clock, it will be less than what is expected.  

cc @mmahadevan108   @hasnainvirk

0xc0170, 2017-10-13:
cc @Sissors - might be able also help with this calculation.

Sissors, 2017-10-16:
That behaviour sounds wrong, so I quickly tested the function, and indeed for 8MHz it returns 6MHz as setting. And this is correct. 

The bus clock is 24MHz. Minimum divisor is 2, so we end up at 12MHz. This we can prescale either to 6MHz or 12MHz. The code is designed never to give a higher frequency than set, so then the closest one is 6MHz. 

Only question is why your 16MHz gets scaled to 8MHz, that should not be possible and it should be 12MHz. 12MHz would be really on the edge for my logic analyzer, and it isn't easily connected to my KL25Z. So before I try that, are you really sure it is running at 8MHz? Also according to the datasheet it should not be able to generate that.

deepikabhavnani, 2018-12-28:
> Only question is why your 16MHz gets scaled to 8MHz, that should not be possible and it should be 12MHz. 12MHz would be really on the edge for my logic analyzer, and it isn't easily connected to my KL25Z. So before I try that, are you really sure it is running at 8MHz? 

@maclobdell - Please verify if the frequency is 12Mhz or 8Mhz?

maclobdell, 2019-01-03:
@ARMmbed/team-nxp  this is a really old issue.  Its a shame we didn't get this resolved.  I take part of the blame on that.  @mmahadevan108 Can you look into this?  requires a logic analyzer connected to the spI clock pin.

mmahadevan108, 2019-01-03:
@maclobdell Looking at the divider values it is not possible to get 8MHz from the 24MHz Bus clock. I do not have the bandwidth to investigate this further.

deepikabhavnani, 2019-01-03:
@mmahadevan108 - Yes for 8MHz selection 6Mhz clock was measured. But we are not sure why 8Mhz clock was noted when 16Mhz was selected, it should be 12Mhz as per calculations

closed 2021-10-20 by ciarmcom