ARMmbed/mbed-os #5048: "Malformed input " messages while compiling WiFi example

deepikabhavnani, 2017-09-07:
Compiling Mbed-os WiFi example throws warnings as shown below, when compiled for "REALTEK_RTL8195AM". 

Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(rtk_wlan_if.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(rtw_psk.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(rtw_xmit.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(rom_rtw_message.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(lxbus_ops.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(freertos_ioctl.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(halhwimg8195a_bb.o)
Malformed input found when parsing ARMCC map: lib_wlan_mbed_arm.ar(Hal8195APwrSeq.o)

Steps to reproduce:
1. mbed compile -t ARM -m REALTEK_RTL8195AM

RonEld, 2017-09-14:
I reproduced this issue on a different board. It hapopens when building with a .ar file as input.
It's when parsing the map file in the [following](https://github.com/ARMmbed/mbed-os/blob/master/tools/memap.py#L272) script:
~~~
 RE_OBJECT_ARMCC = r'(.+\.l)\((.+\.o)\)'
            test_re_obj_name = re.match(RE_OBJECT_ARMCC, line)

            if test_re_obj_name:
                object_name = test_re_obj_name.group(1) + '/' + \
                              test_re_obj_name.group(2)

                return '[lib]/' + object_name
            else:
                print "Malformed input found when parsing ARMCC map: %s" % line
                return '[misc]'
~~~
The script `memmap.py` searches for .o files and libraries in the form of .l files.
Since the input is .ar file, this message is printed. Note this is only on ARMCC ( I haven't checked IAR) and it doesn't affect the resulting ELF file.

theotherjimmy, 2017-09-14:
Thanks for the detailed explanation @RonEld. Do you have a recommendation for correcting this? could we just add `.ar` to the list of matched extensions?

RonEld, 2017-09-14:
Hi @theotherjimmy yes.
I did a quick and dirty fix:
~~~
RE_OBJECT_ARMCC = r'(.+\.l)\((.+\.o)\)'
            test_re_obj_name = re.match(RE_OBJECT_ARMCC, line)

            if test_re_obj_name:
                object_name = test_re_obj_name.group(1) + '/' + \
                              test_re_obj_name.group(2)

                return '[lib]/' + object_name
            else:
                RE_OBJECT_ARMCC = r'(.+\.ar)\((.+\.o)\)'
                test_re_obj_name = re.match(RE_OBJECT_ARMCC, line)

                if test_re_obj_name:
                    object_name = test_re_obj_name.group(1) + '/' + \
                                  test_re_obj_name.group(2)

                    return '[lib]/' + object_name
                else:
                    print "Malformed input found when parsing ARMCC map: %s" % line
                    return '[misc]'
~~~
But I believe we can have a better solution, with the regular expression seraching for ` r'(.+\.l)\((.+\.o)\)' ` or ` r'(.+\.ar)\((.+\.o)\)'` in a single line

theotherjimmy, 2017-09-14:
Yeah, I'll code that up. Just a sec

theotherjimmy, 2017-09-14:
Does this patch work for you?
```diff
diff --git a/tools/memap.py b/tools/memap.py
index e425afe10..d847e27fa 100644
--- a/tools/memap.py
+++ b/tools/memap.py
@@ -269,7 +269,7 @@ class MemapParser(object):
 
         else:
 
-            RE_OBJECT_ARMCC = r'(.+\.l)\((.+\.o)\)'
+            RE_OBJECT_ARMCC = r'(.+\.(l|ar))\((.+\.o)\)'
             test_re_obj_name = re.match(RE_OBJECT_ARMCC, line)
 
             if test_re_obj_name:
```

theotherjimmy, 2017-09-14:
It works for me 😄

theotherjimmy, 2017-09-14:
It also moves the library from `[misc]` to it's name!

RonEld, 2017-09-14:
Yes, this patch works for me as well

theotherjimmy, 2017-09-14:
Awesome. I'll make the PR momentarily.

yennster, 2017-09-15:
When compiling for Odin, I receive a huge list of errors all similar to the following output:

```
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_wlan_os_port.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_supplicant.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_hci.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_bt_serial.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_bt_serial_le.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(controller_startup_WL18.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(hci_transport_H4_uart_WL18.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_button.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_powerman.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_sdio.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_wpa.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(hci.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(hci_data.o)
Malformed input found when parsing ARMCC map: libublox-odin-w2-driver.ar(cb_mmc.o)
```

After running `mbed compile --clean -t ARM -m UBLOX_EVK_ODIN_W2 -f` for the `mbed-os-example-wifi` on the mbed-os-5.6-oob branch

theotherjimmy, 2017-09-15:
And does the aforementioned PR fix them for you?

yennster, 2017-09-15:
@theotherjimmy yes it does!

PR #5103 (2017-09-18): Parse libraries with memap-arm
Resolves #5048

Thanks @RonEld!
closed 2017-09-18 by theotherjimmy