ARMmbed/mbed-os #5051: us_ticker_fire_interrupt: Event handler callback is never called

fkjagodzinski, 2017-09-08:
### Description
- Type: Bug
- Priority: Major

## Bug

**Target**
NUCLEO_F070RB, EFM32GG_STK3700

**Toolchain**
GCC_ARM, ARM, IAR

**Toolchain version**
GCC_ARM _v6.3.1_,
ARM _uvision v5.24.2.0,_
IAR _IAR ANSI C/C++ Compiler V7.80.4.12462/W32 for ARM_

**mbed-cli version** 
1.2.0

**mbed-os sha**
814eaa99f

**Code**
```cpp
#include "mbed.h"
#include "hal/ticker_api.h"
#include "rtos.h"

Semaphore sem(0,1);

void foo(uint32_t id) {
    (void) id;
    sem.release();
}

int main() {
    ticker_event_t e;
    const ticker_data_t *us_td = get_us_ticker_data();
    ticker_set_handler(us_td, foo);
    ticker_insert_event_us(us_td, &e, 0, NULL);

    // As ticker_insert_event_us() is called with a timestamp earlier
    // than the current  -- us_td->interface->fire_interrupt is called.

    // sem should have been released by us_ticker_fire_interrupt()
    int32_t num_tokens = sem.wait(0);

    DigitalOut led1(LED1, 1);
    while (true) {
        if (num_tokens) {
            led1 = !led1;
        }
        wait(0.1);
    }
}
```

**Expected behavior**
The ticker event handler `foo()` is called immediately after `ticker_insert_event_us()`.

**Actual behavior**
The ticker event handler `foo` is never called.

cc @bulislaw @0xc0170

bulislaw, 2017-09-08:
So this change is incorrect? https://github.com/ARMmbed/mbed-os/pull/5019/files

If not all targets conform we shouldn't change documentation. We could add a check in the function to actually not insert the event if it was in the past. We can fix it when we will come to rewriting HAL spec for that.

fkjagodzinski, 2017-09-08:
I'd say that #5019 is correct because an event _is inserted_ for every target I've tested.

This issue regards what happens just after the insertion. The event handler isn't called on NUCLEO_F070RB & EFM32GG_STK3700. Other boards like K64F, NUCLEO_L476RG and NUCLEO_F429ZI work as expected -- the handler is called immediately after insertion of the event.

bulislaw, 2017-09-08:
But the original specification said it won't be called, so actually only these boards conform? Am I missing something?

0xc0170, 2017-09-08:
`    ticker_insert_event_us(us_td, &e, 0, NULL);` would be better to get current and use current - some margin (`0` should work also though).

@fkjagodzinski Is this an issue with fire interrupt? I quickly checked EFM32, seems like we might have missed to set overflow counter to 0 if fire_interrupt is invoked.

@stevew817 @akselsm 

@bcostm @adustm @LMESTM @jeromecoutant

fkjagodzinski, 2017-09-08:
> the original specification said it won't be called

@bulislaw Please take a look at [`/TESTS/mbed_hal/ticker/main.cpp`](https://github.com/ARMmbed/mbed-os/blob/5bddd881e98101f5c491cccf3367894ecb4e89dd/TESTS/mbed_hal/ticker/main.cpp#L941).

fkjagodzinski, 2017-09-14:
The issue seems to be resolved for **NUCLEO_F070RB** board by #5012.

For the **EFM32GG_STK3700** however -- the bug is still present.

0xc0170, 2017-09-22:
> For the EFM32GG_STK3700 however -- the bug is still present.
 
bump @stevew817 @akselsm

0xc0170, 2017-09-22:
As suspected overflow counter and flag needed to be set. Fixed here: https://github.com/ARMmbed/mbed-os/pull/5176

fkjagodzinski, 2017-09-25:
#5176 fixes this issue for **EFM32GG_STK3700**.

fkjagodzinski, 2017-09-26:
Reopening -- need to wait for #5176.

PR #5176 (2017-09-29): EFM32: fix fire interrupt (set flags)
There's overflow counter that needs to be 0, and
CC0 flag set.

Fixes #5051

Tested with example in the issue, the callback is fired as it should. I've been testing it with leopard gecko (any efm32 should do the trick).

cc @fkjagodzinski
commit 29988933 (2017-09-29): EFM32: fix fire interrupt (set flags)

There's overflow counter that needs to be 0, and
CC0 flag set.

Fixes #5051
closed 2017-09-29 by theotherjimmy