ARMmbed/mbed-os #5054: REALTEK_RTL8195AM: 'mbed compile' for cloud client example with ARM toolchain results to link errors

JammuKekkonen, 2017-09-08:
**Target**
REALTEK_RTL8195AM

**Toolchain:**
ARM

**mbed-os sha:**
e12f116ec

**Expected behavior**
mbed compile command goes through and produces bin file

**Actual behavior**
mbed compile gets 33 errors during linking.

**Steps to reproduce**
- checkout branch 'REALTEK_RTL8195AM_support' from mbed-cloud-client-example-internal
- run 'mbed compile -m REALTEK_RTL8195AM -t ARM'

0xc0170, 2017-09-08:
@Archcady

0xc0170, 2017-09-08:
@JammuKekkonen what are the linker errors? Can you paste the output from the console please?

JammuKekkonen, 2017-09-08:
Link: mbed-cloud-client-example-internal
Error: L6218E: Undefined symbol NS_CMSG_NXTHDR (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol ns_in6addr_any (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol socket_close (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol socket_open (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol socket_recvmsg (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol socket_sendmsg (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol socket_setsockopt (referred from coap_connection_handler.o).
Error: L6218E: Undefined symbol arm_net_address_get (referred from ethernet_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_interface_configure_ipv6_bootstrap_set (referred from ethernet_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_interface_down (referred from ethernet_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_interface_ethernet_init (referred from ethernet_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_interface_up (referred from ethernet_tasklet.o).
Error: L6218E: Undefined symbol ethernet_mac_create (referred from ethernet_tasklet.o).
Error: L6218E: Undefined symbol net_init_core (referred from mesh_system.o).
Error: L6218E: Undefined symbol arm_net_get_current_channel (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_6lowpan_link_nwk_id_filter_for_nwk_scan (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_6lowpan_link_panid_filter_for_nwk_scan (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_6lowpan_link_scan_parameter_set (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_interface_configure_6lowpan_bootstrap_set (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_interface_lowpan_init (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_link_layer_security_mode (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_mac_address_read (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_nd_address_read (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_set_channel_list (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol ns_sw_mac_create (referred from nd_tasklet.o).
Error: L6218E: Undefined symbol arm_nwk_host_mode_set (referred from thread_tasklet.o).
Error: L6218E: Undefined symbol thread_management_node_init (referred from thread_tasklet.o).
Error: L6218E: Undefined symbol socket_accept (referred from NanostackInterface.o).
Error: L6218E: Undefined symbol socket_bind (referred from NanostackInterface.o).
Error: L6218E: Undefined symbol socket_connect (referred from NanostackInterface.o).
Error: L6218E: Undefined symbol socket_getsockopt (referred from NanostackInterface.o).
Error: L6218E: Undefined symbol socket_listen (referred from NanostackInterface.o).
Error: L6218E: Undefined symbol socket_recvfrom (referred from NanostackInterface.o).
Finished: 0 information, 0 warning and 33 error messages.

0xc0170, 2017-09-08:
mbed-os-example-blinky builds for this target, with ARM toolchain.

This is related to nanostack, it seems. Something not correctly set?
@SeppoTakalo

SeppoTakalo, 2017-09-12:
Error reported here does not make sense.
Yes, those mentioned symbols are all related to Nanostack. And that is the problem.

[The default](https://github.com/ARMmbed/mbed-cloud-client-example-internal/blob/REALTEK_RTL8195AM_support/mbed_app.json) `mbed_app.json` does not specify `FEATURE_NANOSTACK` and therefore this cannot happen in cleanly cloned environment.

This can only happen in you specify `NANOSTACK` in the `"target.features_add"` line without specifying any of the binary targets of Nanostack. See https://github.com/ARMmbed/Handbook/blob/new_engine/docs/refs/mesh/quick_start_build.md

JammuKekkonen, 2017-09-13:
@SeppoTakalo that's right, the nanostack feature was added on the build. What confused me here was the fact that gcc_arm build compiled cleanly. It turns out that gcc_arm and arm target linkers work differently; gcc_arm will not fail on missing refs on stuff that gets optimized out anyways due to not being used.
Closing non-issue.

closed 2017-09-13 by JammuKekkonen

SeppoTakalo, 2017-09-13:
Yes, true.
IAR and GCC are able to link files with unresolved dependencies, if those references are not used. Basically they link in a mode where they append only referenced symbols into the image and stop once all references are in.
ARM seems to work in reverse way where it resolves all dependencies before trying to link anything.