ARMmbed/mbed-os #5056: REALTEK_RTL8195AM: mbed-cloud-client-example-internal gcc_arm build unstable

JammuKekkonen, 2017-09-08:
**Target**
REALTEK_RTL8195AM

**Toolchain:**
GCC_ARM

**mbed-os sha:**
e12f116

**Expected behavior**
mbed cloud client example is able to register to server and run

**Actual behavior**
running the application hangs some time after connecting to wifi access point. without traces the app does not get to registered state. with traces enabled (changing config override 'mbed-trace.enable' value to 1), the registration sometimes goes through but the app still stalls after a while.

**Steps to reproduce**
- checkout branch 'REALTEK_RTL8195AM_support' from mbed-cloud-client-example-internal
- compile with GCC_ARM toolchain
- flash&run the app

0xc0170, 2017-09-08:
cc @Archcady @JanneKiiskila

0xc0170, 2017-10-13:
@JanneKiiskila @JammuKekkonen Is this still true ? There have been some fixes for 5.6.

marcuschangarm, 2017-10-13:
Yes, there are still some issues.

samchuarm, 2017-12-14:
Hi @JammuKekkonen Can this issue still be reproduced with the latest master? There's been several fixes from Realtek to address instability since Oct.

Archcady, 2017-12-20:
```
=========================================================

ROM Version: 0.3

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003)

=========================================================
Check boot type form eFuse
SPI Initial
Image1 length: 0x2ea8, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1

Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (RTW)
[EasyConnect] Connecting to WiFi IOT_MERCURY_59E0
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 00:e0:4c:87:00:00
[EasyConnect] IP address 192.168.90.101

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
handle_timer_tick, device not registered
handle_timer_tick, device not registered

Registered object successfully!
handle_timer_click, new value of counter is 1
handle_timer_click, new value of counter is 2
handle_timer_click, new value of counter is 3
handle_timer_click, new value of counter is 4
handle_timer_click, new value of counter is 5
handle_timer_click, new value of counter is 6
handle_timer_click, new value of counter is 7
handle_timer_click, new value of counter is 8
handle_timer_click, new value of counter is 9
handle_timer_click, new value of counter is 10
handle_timer_click, new value of counter is 11
handle_timer_click, new value of counter is 12
handle_timer_click, new value of counter is 13
handle_timer_click, new value of counter is 14
handle_timer_click, new value of counter is 15
handle_timer_click, new value of counter is 16
handle_timer_click, new value of counter is 17
handle_timer_click, new value of counter is 18
handle_timer_click, new value of counter is 19
handle_timer_click, new value of counter is 20
handle_timer_click, new value of counter is 21
handle_timer_click, new value of counter is 22
handle_timer_click, new value of counter is 23

```

The connection to mbed clinet is stable with the latest master, can we close this issue?

closed 2017-12-20 by 0xc0170