ARMmbed/mbed-os #5059: STM32 CAN bus

pilotak, 2017-09-08:
I tried CAN bus sucessfully on LPC1768, now i moved to NUCLEO_F142ZG and NUCLEO_F103RB but i can't get it to work. Same code works on LPC but doesn't on STM32. Is CAN not supported on STM? I looked at transmit pin through scope and there is nothing = pin HIGH.

```cpp
#include "mbed.h"
CAN can1(PB_8, PB_9);
DigitalOut led1(LED1);
char counter = 0;

int main() {
    printf("start\n");
    
    can1.frequency(250000);
    can1.reset();
    can1.mode(CAN::Normal);

    while(1) {
        if(can1.write(CANMessage(1337, &counter, 1))) {
            printf("sent\n");
        } 
        else {
            printf("error\n");
        }
        counter++;
        led1 = !led1;
        
        wait_ms(100);
    }
}
```

It prints 3x "sent" (but not visible on scope) and then error forever

### Description
- Type: Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
**Target**
STM32

**Toolchain:**
GCC_ARM & online

LMESTM, 2017-09-11:
Hi @pilotak  
You get an error after 3 messages being sent because there are 3 transmit mailboxes in STM32 F103RB and none of them is free. So it seems like the transmission is failing.  How have you connected the board to the CAN bus ? Do you have tranceivers in place like suggested here https://developer.mbed.org/handbook/CAN ?

pilotak, 2017-09-11:
Hi @LMESTM ,
As i already explained , it works perfectly on LPC1768 so yes it tranceivers are working and they are connected as suggested. I also tried to swap RXD/TXD just in case but it's the same - no activity on those two pins, checked with scope.

> I did test on LPC: when i disconnect transmitter from LPC i can still measure activity on TXD pin - it tries to send data.

F412ZG and F103RB there is no activity, even that program sends me `sent` (which should be confirmation that it arrived to the mailbox) there is no activity on TXD pin. When i press restart button on nucleo RXD and TXD gets **LOW** for as long as button is pressed, otherwise **HIGH** all the time

LMESTM, 2017-09-12:
@pilotak 
I think I could reproduce the issue. 
There is a recent fix on can reset that has not been merged yet #4932 
until this fix is merged, the reset is not well managed. 
Can you try one of 2 possibilities:
- either cherry-pick fix from #4932 
- or remove the call to can1.reset() from your example as this is not needed. 
Let me know if any of the 2 above proposals work ok. 
If this is ok, we just have to wait #4932 to be merged.

pilotak, 2017-09-12:
@LMESTM 
by removing reset()  it works. Thanks a lot

> I tried to checkout your PR #4932 which didn't work for me, but i'm sure it works i just didn't want to spend time finding why (because removing one line from code is faster :bowtie:)

I guess we will have to wait until merge to properly work.

closed 2017-09-12 by pilotak

LMESTM, 2017-09-12:
Thanks for your quick feedback - glad it works.
When #4932 is merged, you can re-introduce the use of reset in your code (if/when reset is needed ... ;-) )

pilotak, 2017-09-21:
just for reference, it has been merged

amrithmmh, 2018-09-28:
will this work on stm32f103ct6 ? I meant the default CAN api in mbed

pilotak, 2018-09-28:
should, you can try