ARMmbed/mbed-os #5064: Condition variable

DuyTrandeLion, 2017-09-09:
## Question

I am working on mbed RTOS and I realized that there is no document about condition variable. Do you support or is there any document about how to implement condition variable in mbed os? I tried searching but there was almost no document about it. 

Thank you very much.

geky, 2017-09-11:
Good thing to notice

@c1728p9 put up a pr to add a ConditionalVariable class, though it ended up dropped: https://github.com/ARMmbed/mbed-os/pull/3648

For most cases, condition variables can be replaced with the EventFlags class ([here](https://github.com/ARMmbed/mbed-os/blob/master/rtos/EventFlags.h#L43)), although synchronization needs to be handled seperately. The documentation should be updated with the 5.6 release which is on the way, but for now there is just the doxygen in the header file.

Here's a simple example:
``` cpp
EventFlags flags;
Mutex mutex;
bool condition = false;

void wait_for_condition() {
    mutex.lock();
    while (!condition) {
        mutex.unlock();
        flags.wait_any(1);
        mutex.lock();
    }
    mutex.unlock();
}

void trigger_condition() {
    mutex.lock(); // note, we don't need this mutex if we are in interrupt context
    condition = true;
    flags.set(1);
    mutex.unlock();
}
```

Here's a real world example where we use event flags as condition variables in the socket classes:
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/UDPSocket.cpp#L76

DuyTrandeLion, 2017-09-12:
Thank you very much for you reply, I guess EventFlags is good as a replacement for ConditionalVariable. Still hope that one day ConditionalVariable would be reopened

kjbracey-arm, 2017-11-14:
#3648 has now been merged, so `rtos::ConditionVariable` is present on master, and will be in mbed-os-5.7.

closed 2017-11-14 by 0xc0170