ARMmbed/mbed-os #5067: Deep sleep lock would overflow (> USHRT_MAX)

betzw, 2017-09-11:
### Description
- Type: Bug
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**mbed-os sha:**
3f2e986 Merge pull request #5012 from LMESTM/issue5004_ticker_16b

**Actual behavior**
See [stack_tasks.txt](https://github.com/ARMmbed/mbed-os/files/1292547/stack_tasks.txt)

betzw, 2017-09-11:
Maybe class `Timer` requires a destructor now?

0xc0170, 2017-09-22:
This should be fixed by https://github.com/ARMmbed/mbed-os/pull/5063. The issue was actually with Callbacks operato== (also fixed now on master). I'll close this, reopen with new findings if any

Thank you for reporting

betzw, 2017-09-22:
Unfortunately #5063 does **not** fix this issue!
Whereas modifications like in this [diff](https://github.com/ARMmbed/mbed-os/files/1324633/diff.txt) seem to fix my issue (see also my [comment](https://github.com/ARMmbed/mbed-os/issues/5067#issuecomment-328522178)  above).

betzw, 2017-09-22:
@0xc0170, could you pls. re-open this issue.

0xc0170, 2017-09-22:
OK, going to check

0xc0170, 2017-09-22:
`Timer` has a problem with destructor - it is never stoped:

```
{
  {
    Timer timer;

    timer.start();
  }
// here timer object is out of scope, but it's running as nothing stopped it - it assumes us ticker is always running
}
```

Thus would be good to to define dtor with stoping the timer (unblock deepsleep if us ticker is involved).

SerialBase has an empty virtual dtor. It should not be empty in this case - assuming either attach or even async is invoked, then destroying the object does not clean its resources (no abort, or detach is involved).

We will propose a fix for this

cc @c1728p9

screamerbg, 2017-09-28:
@0xc0170 @c1728p9 any idea about providing a fix for this? This is gating the WiFi driver @betzw is working on.

0xc0170, 2017-09-28:
There's one improvement here https://github.com/ARMmbed/mbed-os/pull/5148 (not yet fixing it).  I'll create another PR to create Timer dtor as proposed above based on this improvement (will need it).

Regarding SerialBase - I'll talk to @c1728p9

c1728p9, 2017-09-28:
Hi @betzw, it looks like one of the fixes in the diff you provided has already been made in #5095. I created #5220 which fixes the other problem shown in your diff (along with other potential problems). 

@betzw can you try #5220 to confirm it fixes your problem?

betzw, 2017-09-29:
I will try asap, but most likely today I won't have the time ... :frowning_face:

betzw, 2017-10-03:
Yep, https://github.com/c1728p9/mbed-os/tree/sleep_lock_fixes seems to fix this issue.

closed 2017-10-03 by betzw