ARMmbed/mbed-os #5068: SDBlockDevice init failure on mbed-os-5.5

costanic, 2017-09-11:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
K64F and UBLOX EVK-ODIN-W2

**Toolchain:**
GCC_ARM

**Toolchain version:**
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

**mbed-cli version:** 
$ mbed --version
1.2.0

**mbed-os sha:**
$ git log -n1 --oneline
db4be9469 (HEAD, tag: mbed_lib_rev149, tag: mbed-os-5.5.5) Merge pull request #4889 from ARMmbed/release-candidate

Also fails with:
* 4c256f045 (HEAD, tag: mbed_lib_rev148, tag: mbed-os-5.5.4) Merge pull request #4824 from ARMmbed/release-candidate
* 98ba8acb8 (tag: mbed_lib_rev150, tag: mbed-os-5.5.6, tag: latest, origin/mbed-os-5.5) Merge pull request #4974 from ARMmbed/release-candidate

** sd-driver sha:**
$ git log -n1 --oneline
2d24758 (HEAD, tag: sd-driver-0.1.1-mbed-os-5.5.4) Merge pull request #32 from ARMmbed/Readme_update

Also fails with:
$ git log master -n1 --oneline
9edd43d (origin/master, origin/HEAD, master) Merge pull request #60 from ARMmbed/g-fix-travis

**Expected behavior**
sd.init() succeeds

**Actual behavior**
sd.init() returns failure code -5005

**Steps to reproduce**
Follow instructions at https://github.com/ARMmbed/sd-driver/blob/9edd43dba50b861d0928ceda9a60bbd0e1f96d36/README.md#sdblockdevice-example-application

Compile the application for either K64F or UBLOX EVK-ODIN-W2, the issue reproduces on both platforms.

** Additional notes **
1. This was tried with two SD cards, both failed:
* SanDisk Edge 4GB Micro SDHC
* SanDisk 8GB Micro SDHC

2. This issue does not occur on older mbed-os-5.4.7 with older sd-driver
This combination works:
mbed-os: fc1836545 (HEAD, tag: mbed_lib_rev144, tag: mbed-os-5.4.7, origin/mbed-os-5.4) Merge pull request #4463 from ARMmbed/release-candidate
sd-driver: $ git log -n1 --oneline
efc6bbc (HEAD) Add pin override for NUCLEO_F429ZI (#20)

3. Re-format the SD card and power-cycle the device when switching library versions to increase chances of success.  Even with the working driver combination, the init would sometimes fail the first few times until a power cycle.

4. Marked as a blocker since any FOTA with mbed-os-5.5.4/mbed-cloud-client-1.2.2-EA will fail when configured to use SD card for file storage.

deepikabhavnani, 2017-09-12:
Initialization and some other issues with SD cards were resolved in https://github.com/ARMmbed/sd-driver/pull/61. Please try the SD-driver master (#1cb007fb3809a5e2d56d179e81332615c0a9b9fd)

costanic, 2017-09-12:
confirmed, init works with the following combination:

**mbed-os:**
commit 4c256f04596179699c4f14b6863b07cc024ca9be (HEAD, tag: mbed_lib_rev148, tag: mbed-os-5.5.4)
Merge: ed9d1da9d c511e96e0
Author: Jimmy Brisson <theotherjimmy@gmail.com>
Date:   Wed Aug 2 15:45:49 2017 -0500

    Merge pull request #4824 from ARMmbed/release-candidate

    Release candidate for mbed-os-5.5.4

**sd-driver:**
commit 1cb007fb3809a5e2d56d179e81332615c0a9b9fd (HEAD -> master, origin/master, origin/HEAD)
Merge: 9edd43d c46d077
Author: Deepika <deepika.bhavnani@arm.com>
Date:   Mon Sep 11 11:36:31 2017 -0500

    Merge pull request #61 from deepikabhavnani/sd_diff_cards

    Deselecting card between read/write ops caused issues

closed 2017-09-12 by costanic