ARMmbed/mbed-os #5070: Blinky doesn't work when using develop/release profile with external power

MarceloSalazar, 2017-09-11:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/issues/4686
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
(u-blox-EVK-NINA-B1)[https://developer.mbed.org/platforms/u-blox-EVK-NINA-B1]

**Toolchain:**
GCC_ARM

**gcc 6.3.1**

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
mbed OS 5.5.6 (98ba8acb83cfc65f30a8a0771a27c71443ab093a)

**DAPLink version:**
243

**Expected behavior**
Blinky should just work fine with all profiles (debug, release, develop), regardless of whether you use the USB port or external power supply.

**Actual behavior**
Blinky doesn't work when using the develop and release profiles with external power supply of the NINA B1 platform. It does work well with the debug profile.
It seems the system goes to sleep and never wakes up.

This has been found when working with @NeilMacMullen and @pan-

jrobeson, 2017-09-13:
@MarceloSalazar : can you try the current master branch of mbed-os?

MarceloSalazar, 2017-09-14:
I've tried [Blinky](https://github.com/ARMmbed/mbed-os-example-blinky/tree/mbed-os-5.6-oob) with [mbed-os-5.6.0-rc1](https://github.com/ARMmbed/mbed-os/commit/a4056fb8bf3ee10b8e037d332f8fbd0c68169ef1) and can see this working on NINA when using release/develop profiles.

@NeilMacMullen could you give it a try on your custom platform and confirm?

pan-, 2017-09-15:
#5086 will also help new targets to avoid this issue in the future.

MarceloSalazar, 2017-09-28:
Confirmed it's solved.

closed 2017-09-28 by MarceloSalazar