ARMmbed/mbed-os #5072: Serial communication not working with Flowcontrol disabled on NRF51 devices.

SenRamakri, 2017-09-11:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major 
---------------------------------------------------------------
## Serial communication not working with Flowcontrol disabled on NRF51 devices.

**Target**
NRF51_DK | Similar devices

**Toolchain:**
GCC_ARM|ARM|IAR

**Steps to reproduce**

Run the below application on a NRF51_DK and and observe that Serial echo doesn't work.

#include "mbed.h"

int main()
{
  Serial *pc = new Serial(USBTX, USBRX);
  pc->format(8, Serial::None, 1);
  pc->set_flow_control(Serial::Disabled);
  pc->baud(115200);
  pc->printf("LOOPBACK SERIAL  with callback>>>>>>\r\n");

  while (1)
  {
    while (pc->readable())
    {
      char c = pc->getc();
      pc->putc(c); pc->putc(' ');
    }
    Thread::wait(10);
  }
}

We see that on these devices unless we program PSELRTS/PSELCTS registers with the right values RX/TX tasks doesn't work when Flowcontrol is disabled. We are currently pushing a workaround for this issue.
https://github.com/ARMmbed/mbed-os/pull/5071

Please review if this fix is feasible.

pan-, 2017-09-12:
@nvlsianpu Andrzej, could you have a look at this issue, I've observed the same on NRF52.

nvlsianpu, 2017-09-13:
@pan- Yes, I will investigate this soon.

nvlsianpu, 2017-09-13:
@pan-, @SenRamakri: Checked - It is Nordic cmsis-dap IF issue. I tested this using external FTDI converter and with j-link IF firmware  as well - loopbacks works well.

Apparently nordic cmsis-dap IF (nRF5x-OB-mbed-IF version 0221, available at http://www.nordicsemi.com/eng/nordic/Products/nRF51-DK/nRF5x-OB-mbed-IF/38673)doesn't serve serial communication without hardware flow control.
I guess that your patch is working because it set expected states on CTS&RTS pins connected to the IF chip

This is not a bug in mbed. I'm going to report this to the nordic IF maintainer.

SenRamakri, 2017-09-13:
Hello @nvlsianpu,
Thanks for looking into this, once we fix the DAP IF issue feel free to remove the workaround I'm pushing now. Or if its going to be done sooner we may not even have to get the workaround in.

donatieng, 2018-07-25:
Closing as it's a DAP issue, feel free to re-open otherwise

closed 2018-07-25 by donatieng