ARMmbed/mbed-os #5075: Example for BLE on ODIN-W2

janjongboom, 2017-09-12:
Could someone ping someone in u-blox, and see if they can publish a BLE example for the ODIN-W2? They ship a binary driver with the board, but no example code. I assume their own SDK examples for BLE work, so maybe publish an mbed program with those loaded.

Preferably they'd adopt to the mbed BLE API, but that's a longer shot.

andreaslarssonublox, 2017-09-25:
@andreaspeterssonublox

0xc0170, 2017-10-13:
Has the example been published or not?

senthilr, 2017-10-27:
Any progress on this? There is quite a bit of interest in the mbed community for BLE on ODIN-W2
@screamerbg

andreaslarssonublox, 2017-11-02:
Hi, we're currently working on the C++ BLE API port so we would recommend to use that API instead of the C API.
Once the port is finished the mbed BLE examples should work.

BruceSHoward, 2018-02-09:
I have an EVK-ODIN-W2 based application that could really benefit from mbed BLE support that makes use of the on-board BLE radio in the ODIN-W2xx module.  Is there an update on the release date for the C++ BLE API port mentioned by andreaslarssonublox?

janjongboom, 2018-02-12:
@andreaslarssonublox ^

andreaslarssonublox, 2018-02-12:
Hi,	
Support for BLE in Arm Mbed OS5 (C++ API) is being integrated. Plan is to provide a first set of functionality in Q1 2018.

BruceSHoward, 2018-02-13:
Hi ,
Is it possible to use mbed 5.x easy-connect with the ODIN-W2 module configured as an access point instead of as a station? If so, would it be possible to change from one mode to the other programmatically?

andreaslarssonublox, 2018-02-22:
Hi, @BruceSHoward access point mode is not yet supported for the C++ Wi-Fi API.

BruceSHoward, 2018-02-23:
Hi Andreas Larsson,
Are there any plans to support access point mode, with the ability to switch between access point and station mode,  in the future?

asifrizwanubx, 2018-09-10:
We have started the work on mbed BLE support in this sprint (as a high priority feature).

u-blox internal ref: TE_OM2-163

AmmadRehmat, 2019-05-23:
@MarceloSalazar  BLE is now supported

0xc0170, 2019-05-23:
Closing as resolved (confirmed the feature is supported)

closed 2019-05-23 by 0xc0170