ARMmbed/mbed-os #5076: Fail to enter hal_deepsleep when LowPowerTimer is used

ccli8, 2017-09-12:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/pull/4912
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472
NUMAKER_PFM_M453
NUMAKER_PFM_M487

**Toolchain:**
GCC_ARM

**Toolchain version:**
GNU Tools for ARM Embedded Processors 5.4.1 20160919

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
4b98566 Merge pull request #4875 from deepikabhavnani/cm_23_support

**Steps to reproduce**
With sleep manager introduced, deep sleep will be locked (`sleep_manager_lock_deep_sleep`) as `Timer` is started. `LowPowerTimer` inherits `Timer` and so use of it will also lock deep sleep. As I know, `LowPowerTimer` (and `LowPowerTicker`/`LowPowerTimeout`) rely on `lp_ticker` and would still work even though system enters deep sleep mode. They shouldn't lock deep sleep.

```
#include "mbed.h"

int main()
{
    static LowPowerTimer timer;
   
    timer.start();
   
    while (true) {
        deepsleep();
    }

    return 0;
}
```

0xc0170, 2017-09-20:
@ccli8 Thanks for the report, we are looking at this

ccli8, 2017-10-12:
Fixed on mbed-os-5.6.2.

closed 2017-10-12 by ccli8