ARMmbed/mbed-os #5079: MD5, SHA1 and SHA256 hardware acceleration not working for STM32F439xI

andresag01, 2017-09-12:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/issues/4928
- Priority: Major

This problem seems to be closely related to https://github.com/ARMmbed/mbed-os/issues/4928, which is a failure in AES hardware acceleration for STM32F439xI family of targets only. The cause of the failures seems to be that the hardware acceleration code for these targets cannot correctly handle interleaved operation of multiple mbedtls contexts for the SHA1, MD5 and SHA256 crypto primitives. I have written the following test that exercises the problem:
```C
#include "mbed.h"

#include "mbedtls/aes.h"
#include "mbedtls/sha1.h"
#include "mbedtls/sha256.h"
#include "mbedtls/des.h"
#include "mbedtls/md5.h"

#include "mbedtls/platform.h"
#include "mbedtls/error.h"

#define REPETITIONS 5

#if !defined(MBEDTLS_AES_ALT)
#warning "AES hardware acceleration is not enabled!"
#endif

#if !defined(MBEDTLS_SHA1_ALT)
#warning "SHA1 hardware acceleration is not enabled!"
#endif

#if !defined(MBEDTLS_SHA256_ALT)
#warning "SHA256 hardware acceleration is not enabled!"
#endif

#if !defined(MBEDTLS_MD5_ALT)
#warning "MD5 hardware acceleration is not enabled!"
#endif

/*****************************************************************************/
/*************************AES-CBC tests***************************************/
/*****************************************************************************/

#define AES_CTX_COUNT 12

const unsigned char keys[AES_CTX_COUNT][16] = {
    { 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf8, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfc, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0xe3, 0x7b, 0x1c, 0x6a, 0xa2, 0x84, 0x6f, 0x6f,
      0xdb, 0x41, 0x3f, 0x23, 0x8b, 0x08, 0x9f, 0x23, },
    { 0x6c, 0x00, 0x2b, 0x68, 0x24, 0x83, 0xe0, 0xca,
      0xbc, 0xc7, 0x31, 0xc2, 0x53, 0xbe, 0x56, 0x74, },
    { 0x14, 0x3a, 0xe8, 0xed, 0x65, 0x55, 0xab, 0xa9,
      0x61, 0x10, 0xab, 0x58, 0x89, 0x3a, 0x8a, 0xe1, },
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, },
};

const unsigned char inputs[AES_CTX_COUNT][16] = {
[... 836 lines cut ...]

    return( exit_code );
}
```

---------------------------------------------------------------
## Bug

**Target**
STM32F439xI family of devices with hardware acceleration enabled

**Toolchain:**
GCC_ARM

**mbed-os sha:**
5e437fe5ea8793d1da81157d3c6c42bdff64e67e

**Expected behavior**
The test should succeed.

**Actual behavior**
The tests for MD5, SHA1 and SHA256 fail

**Steps to reproduce**
Compile and run the test above in a STM32F439xI device, e.g. UBLOX_EVK_ODIN_W2. The failures will be printed in the serial terminal.

andresag01, 2017-09-12:
@RonEld @adustm @yanesca @sbutcher-arm

andresag01, 2017-09-12:
I have created PR https://github.com/ARMmbed/mbed-os/pull/5080 that disables HW acceleration for the relevant primitives. This is only a workaround though.

andresag01, 2017-09-12:
@adustm: My understanding is that the following targets all share the same hardware acceleration code:
* TARGET_STM32F4/
* TARGET_STM32F7/
* TARGET_STM32L4/ 
Do you think they all have similar issues? Unfortunately I do not have the hardware to test them all and report my findings...

adustm, 2017-09-12:
Hello @andresag01 
Thanks for reporting the issue, I'll have to understand the reason for these failure (especially the SHA256 that is not failing with TESTS/mbedtls/multi/main.cpp)

I filnalize the AES then I come back on this one. 

I'll check at the same time the behavior of other families.

kind regards

adustm, 2017-09-12:
By the way, Mihail has all the platforms you are missing :)

andresag01, 2017-09-12:
@adustm:

> Thanks for reporting the issue... I filnalize the AES then I come back on this one

Many thanks for taking a look. Please let me know if there is anything else I can do to help.

> I'll have to understand the reason for these failure (especially the SHA256 that is not failing with TESTS/mbedtls/multi/main.cpp)

Thats a very good point, I had not considered that... My guess is that the test I am currently using is slightly more stressful and uses at least double the amount of contexts. Also, there is a lot of interleaving. Of course, there is also a chance that I made a mistake in the test, but I havent found any indication of this yet.

> Mihail has all the platforms you are missing

Thanks for letting me know! I will speak with Mihail's team and see what I can get

> I'll check at the same time the behavior of other families.

Thanks for checking this. I think that would be ideal given that the code is mostly shared between all the devices. I will also try to get a hold of the boards and run a couple of quick tests myself.

kjbracey-arm, 2017-10-05:
In case it's useful, I've was seeing failures of DTLS/UDP handshakes against mbed Cloud server on the Ublox Odin W2. On 5.6.0, with the crypto acceleration enabled, it fails, but on 5.6.1 with it disabled it works.

But what's odd, and that I can't explain is that TLS/TCP handshakes did actually work on the same system in 5.6.0 with the hardware crypto enabled.

I can't see any difference in what mbed TLS is doing between DTLS and TLS with regards to the calls to crypto. My current best guess is an alignment difference, but I'm far from certain of that. So when you have a fix for the interleaving, I'd be interested in checking it on that DTLS case - I am wondering if the failure cause is something other than interleaving that you've yet to discover.

My second concern is that looking at https://github.com/ARMmbed/mbed-os/pull/5018, I see protection against interleaved crypto operations on a single thread, but I'm not seeing any multithreading protection - doesn't it need a mutex to prevent a second thread touching the hardware between your "restore context" and the actual operation?

You should have a test case that does kind of the same thing as above, but with AES_CTX_COUNT threads.

andresag01, 2017-10-06:
@kjbracey-arm: Thanks for your input. Please note that I have created https://github.com/ARMmbed/mbed-os/issues/5036 to add more stress tests like the one you are suggesting.

adustm, 2017-11-30:
Hello @andresag01 
I'm back on this topic eventually.
I've analysed your issue. 
The problem is not the variety of contexts and the mix of them, the problem is only the 1st input that is empty and the test calls `mbedtls_[sha1/sha256/md5]_finish` without any call to `mbedtls_[sha1/sha256/md5]_update` ( the test goes from `start `to `finish `without `_update`).

The internal state machine of ST HAL code assumes that you've been through it even for an empty buffer.

I can easily fix that, in case any user wants to optimize its application and avoid calling _update in case of empty buffer.

My PR is ready, but I am not able to test MD5. It does not link. 
Can you help me to make it link ?
I've created a `mbedtls-config-changes.h` that contains
`#define MBEDTLS_MD5_C`
I've created a mbed_app.json file with
```
{
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls-config-changes.h\""]
}
```
I've added the test at the root of mbed-os and typed
`mbed compile -m NUCLEO_F439ZI -t GCC_ARM -c -vv`

I get the following link error
```
[DEBUG] Preproc: arm-none-eabi-cpp -E -P .\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F439xI\device\TOOLCHAIN_GCC_ARM\STM32F439ZI.ld -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap
,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -
mfloat-abi=softfp -o .\BUILD\NUCLEO_F439ZI\GCC_ARM\.link_script.ld
[DEBUG] Return: 0
[DEBUG] Link: arm-none-eabi-gcc @.\BUILD\NUCLEO_F439ZI\GCC_ARM\.link_files.txt
[DEBUG] Return: 1
[DEBUG] Errors: ./BUILD/NUCLEO_F439ZI/GCC_ARM/./main.o: In function `md5()':
[DEBUG] Errors: main.cpp:(.text._Z3md5v+0x28): undefined reference to `mbedtls_md5_init'
[DEBUG] Errors: main.cpp:(.text._Z3md5v+0x3a): undefined reference to `mbedtls_md5_starts'
[DEBUG] Errors: main.cpp:(.text._Z3md5v+0x92): undefined reference to `mbedtls_md5_update'
[DEBUG] Errors: main.cpp:(.text._Z3md5v+0xba): undefined reference to `mbedtls_md5_finish'
[DEBUG] Errors: main.cpp:(.text._Z3md5v+0xf0): undefined reference to `mbedtls_md5_free'
[DEBUG] Errors: collect2.exe: error: ld returned 1 exit status
```
How to enable MBEDTLS_MD5_C ?

Kind regards
Armelle

mazimkhan, 2017-12-12:
@adustm can you try un-commenting line https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/inc/mbedtls/config.h#L2046
in file ```mbed-os/features/mbedtls/inc/mbedtls/config.h``` in your workspace.

andresag01, 2017-12-12:
Hi @adustm, my apologies for the delay in my reply. I am not sure if putting the application inside the mbed-os directory works. My understanding is that this directory should only contain the library code and tests, but the stress test that I wrote is written as if it was a user application. To test this, I normally just replace the code in the main.cpp of one of the applications at [mbed-os-example-tls](https://github.com/ARMmbed/mbed-os-example-tls) and then use it as normal. To illustrate this, I have done the setup for you in [here](https://github.com/andresag01/mbed-os-example-tls/tree/md_test) in my fork which is ready to deploy your mbed-os fork and test the change with the normal `mbed deploy && mbed compile -m NUCLEO_F439ZI -t GCC_ARM -c` command.

After trying that I also found that the linking problem persists and tracked it down to a misplaced include in the file md5_alt.h. The following change fixes the problem as `mbedtls/md5.h` also includes `mbedtls/config.h` and that includes the device and example config files which defines MBEDTLS_MD5_C and MBEDTLS_MD5_ALT.

```
diff --git a/features/mbedtls/targets/TARGET_STM/md5_alt.c b/features/mbedtls/targets/TARGET_STM/md5_alt.c
index 328542b..63fb34b 100644
--- a/features/mbedtls/targets/TARGET_STM/md5_alt.c
+++ b/features/mbedtls/targets/TARGET_STM/md5_alt.c
@@ -17,9 +17,10 @@
  *  limitations under the License.
  *
  */
-#if defined(MBEDTLS_MD5_C)
 #include "mbedtls/md5.h"
 
+#if defined(MBEDTLS_MD5_C)
+
 #if defined(MBEDTLS_MD5_ALT)
 #include "mbedtls/platform.h"
```

I hope this helps, and once more, apologies for the delay to reply. Let me know if there are any further questions, but please note that I am currently not directly involved in the development of hardware acceleration APIs so it might take me a little time to reply.

adustm, 2017-12-14:
Thanks @andresag01 . Moving `#if defined(MBEDTLS_MD5_C)` fixes my link problem. I push that in the PR

PR #5630 (2018-01-31): Fix #5079. Support of call to mbedtls_x_finish without calling mbedtls_x_update
## Description
Fix #5079 
Support of call to mbedtls_x_finish without calling mbedtls_x_update.

## Status
**READY**
Before the PR: failure occurs in case of an empty input and mbedtls_xx_update is not called between xxx_start and xxx_finish

After this PR: 
- SHA1 / SHA256 / MD5 hardware acceleration is activated for STM32F439xI
- An empty input can be encrypted correctly even is the call to mbedtls_xxx_update has been omited

~~[] To do: test MD5, as it is not compiled in mbed-os (I could not find a way to activa [... cut]
commit 3250e2d6 (2018-01-31): Fix #5079. Add the support of call to mbedtls_xxx_finish even if mbedtls_xxx_udate
was not called since mbedtls_xxx_start
commit c06a42b0 (2018-01-31): Merge pull request #5630 from adustm/fix5079_sha1_md5_sha256_hwcrypto

Fix #5079. Support of call to mbedtls_x_finish without calling mbedtls_x_update
closed 2018-01-31 by cmonr