ARMmbed/mbed-os #5082: Fix warnings in K64F drivers

SenRamakri, 2017-09-12:
### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
Currently we are seeing lot of build warnings from enumeration definitions in K64F drivers/headers.

**Target**
K64F|??

**Steps to reproduce**
Just do a clean build and you will see lot of warnings from fsl_xxxx.c/h files.

Since this is from K64F SDK, it would be better for NXP team to fix these issues in their SDK. I'm attaching a git patch with some changes to fix this. You may take these changes if its feasible or implement the fixes in a different way. One possible fix would be to change the enumerations to #defines. The current fix in the patch is to selectively apply typecasting to fix the warnings.

Thanks.
[K64F_BuildWarningsFix.zip](https://github.com/ARMmbed/mbed-os/files/1296213/K64F_BuildWarningsFix.zip)

SenRamakri, 2017-09-15:
Add @maclobdell and @mmahadevan108

0xc0170, 2017-09-27:
@mmahadevan108 Can you share these with sdk team? To get some feedback and possibly fixed in the upcoming releases?

kegilbert, 2017-10-13:
A user made an issue recently about warnings in blinky builds for another device. Ran into this on the online compiler when building for K64F (253 warnings). Are there any updates on this?

@0xc0170 
@mmahadevan108

0xc0170, 2018-06-19:
Thanks @SenRamakri for the report. I added this to the #4222 (similar report, earlier issued). This is a duplicate. Please monitor 4222 issue

closed 2018-06-19 by 0xc0170