ARMmbed/mbed-os #5092: Uvision, make_* exporters python traceback on master

theotherjimmy, 2017-09-13:
# Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
# Bug

## Target
Affects all targets

## Toolchain
It looks like it affects uvision, and all make exporters.

## mbed-os sha
7B428916

## Expected behavior

Export succeeds 

## Actual behavior
```
>mbed export -m k64f -i vscode_gcc_arm
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Traceback (most recent call last):
  File "C:\dev\mbed-os\tools\project.py", line 256, in <module>
    main()
  File "C:\dev\mbed-os\tools\project.py", line 252, in main
    build_profile=profile, app_config=options.app_config)
  File "C:\dev\mbed-os\tools\project.py", line 95, in export
    app_config=app_config)
  File "C:\dev\mbed-os\tools\export\__init__.py", line 341, in export_project
    macros=macros)
  File "C:\dev\mbed-os\tools\export\__init__.py", line 207, in generate_project_files
    exporter.generate()
  File "C:\dev\mbed-os\tools\export\vscode\__init__.py", line 29, in generate
    super(VSCode, self).generate()
  File "C:\dev\mbed-os\tools\export\makefile\__init__.py", line 120, in generate
    ctx.update(self.format_flags())
  File "C:\dev\mbed-os\tools\export\makefile\__init__.py", line 140, in format_flags
    for k, v in self.flags.iteritems():
  File "C:\dev\mbed-os\tools\export\exporters.py", line 111, in flags
    for_asm=True)
  File "C:\dev\mbed-os\tools\toolchains\gcc.py", line 161, in get_compile_options
    opts += ['@%s' % self.get_inc_file(includes)]
  File "C:\dev\mbed-os\tools\toolchains\__init__.py", line 831, in get_inc_file
    include_file = join(self.build_dir, ".includes_%s.txt" % self.inc_md5)
AttributeError: 'GCC_ARM' object has no attribute 'inc_md5'
```

**Steps to reproduce**

bentcooke, 2017-09-13:
The make_iar and iar exporters are working.  This only affects uvision, make_armc5 and make_gcc_arm.

theotherjimmy, 2017-09-13:
Thanks for the verification Ben! I did not test make_iar.

PR #5093 (2017-09-14): Disable response files on export
None of the exporters can handle them, so just don't do it

Resolves #5092
closed 2017-09-14 by theotherjimmy