ARMmbed/mbed-os #5101: mbed-os GAP Central and GATT Client support

HITMAnsOFT, 2017-09-14:
### Description
- Type: Question
- Priority: Major
-----------------------------------------------------------------

## Question

Is there a BLE GAP Central and GATT Client support in mbed OS? If yes, is there an example? I need to make a pure central/client device using either nRF51_DK or nRF52_DK target, not interested in switching roles.

HITMAnsOFT, 2017-09-14:
This is the best example I found so far, but it seems to be for mbed 2: https://developer.mbed.org/users/kenjiArai/code/BLE_Uart_Client/ . I'm using it as a reference to implement my own code on mbed OS 5.5.6.

pan-, 2017-09-15:
@HITMAnsOFT You can use our LED example: [here](https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-LED/)

HITMAnsOFT, 2017-09-15:
@pan- I took a quick look and found only GAP peripheral / GATT server code. Did I miss something?

pan-, 2017-09-15:
You are correct, I wanted to link the [LEDBlinker](https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-LEDBlinker/) example.

HITMAnsOFT, 2017-09-15:
Great, thanks!

closed 2017-09-15 by HITMAnsOFT