ARMmbed/mbed-os #5102: I2C STM32 Open Drain Pullup Issue

Fran6Jack, 2017-09-14:
### Description
- Type: Bug
- Related issue: I2C 5V tolerant I/O could be damaged
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32 (Multitech mDot)

**Toolchain:**
mBed

**Expected behavior**
I2C bus are usually 5V tolerant on all STM32 processor.  If an external device on the I2C bus requires 5V operation, we usually acheive it by using 5V external pull-ups on the bus.  Since signaling uses open-drain output on I2C for both signal SCL and SDA any 5V tolerant MCU will work on a 5V I2C bus

**Actual behavior**
Currently the I2C initialization setup the SDA/SCL pin in pin_mode(sda/scl, OpenDrainPullup) which cause the internal pull-up to be activated by default on OpenDrain output.
Having pull-up activated on a 5V externally pull bus, cause the pin to clamp on the STM32 die diode and could damage the IC (There is a note in STM32 datasheet specifying this issue).
It is understood by all the community that I2C bus should always be externally pulled by physical resistor.  I2C initialization should then be **ALWAYS** OpenDrainNoPull by default.

If a feature of a I2C open drain pull-up by the MCU is required, then a new method should be added in the I2C class to add the **mode** option for OpenDrainNoPull (default) and OpenDrainPullup.

**Steps to reproduce**
Use external pull-up of 4K7 to 5V on each SCL/SDA pin.  Initialize the I2C instance.
Voltage on both pins should be between 3.7-3.9V instead of 5V.
Toggle SCL/SDA pin back in DigitalIn with no Pullup, signal will go back to 5V.

LMESTM, 2017-09-14:
@Fran6Jack thank you for reporting the issue and for detailed explanation. I will come back to you in very short term and correct code as needed.

LMESTM, 2017-09-14:
@Fran6Jack I confirm that your analysis is right. I will change the default settings to no pull. Before delivering the Pull Request I will check the possible impacts on the existing test suite. When testing is complete, I will add the referent to the corresponding Pull Request here. Do you need a patch in the meantime ?

Fran6Jack, 2017-09-14:
@LMESTM A patch would be nice.  I just don't know how do we manage the deployment of it. I'm a little bit newbie on mBed developper environment.  Right now, I'm sending a HAL pin_mode(I2C_SDA, OpenDrainNoPull); before each i2c.write transaction which is not really clean.

LMESTM, 2017-09-15:
@Fran6Jack  If you're not familiar with applying patches, it might be simpler to wait for the next official release. Your solution, even though not efficient, is fine for now I think. 
In case you feel like trying out a change in mbed-os source  (mbed-dev library in online compiler), you can check the change I will propose here: https://github.com/LMESTM/mbed/commit/076b096efe3a7ef363719e1c1f1cdcd63b514c9d
I've started the tests run ... will keep you posted.

LMESTM, 2017-09-15:
@0xc0170   What do you think about below proposal ?

> If a feature of a I2C open drain pull-up by the MCU is required, then a new method should be added in the I2C class to add the mode option for OpenDrainNoPull (default) and OpenDrainPullup.

Could this be a change request to I2C driver ? That would allow test mode like master / slave tests between 2 boards without external pull-up resistors.

0xc0170, 2017-09-19:
> Could this be a change request to I2C driver ? That would allow test mode like master / slave tests between 2 boards without external pull-up resistors.

cc @bulislaw @c1728p9

bcostm, 2017-10-16:
Can you please close this issue ? PR #5113 has been merged) ? Thanks.

ST_TO_BE_CLOSED

closed 2018-01-24 by 0xc0170