ARMmbed/mbed-os #5112: [OOB 5.6] ARMC6 build does not startup on NRF52_DK target

toyowata, 2017-09-15:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

On NRF52_DK target, mbed-os-exampl-blinky does not work by ARMC6 toolchain build.

**Target**
NRF52_DK

**Toolchain:**
ARMC6

**Toolchain version:**
Product: MDK Professional 5.24
Component: ARM Compiler 6.7
Tool: armclang [5c9efc00]

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
mbed-os (a4056fb8bf3e)

**DAPLink version:**
nrf51822aa_atsam3u2c_if_mbed_V0219_02022016.bin

**Expected behavior**
mbed-os-exampl-blinky should blink.

**Actual behavior**
mbed-os-exampl-blinky does not blink.

**Steps to reproduce**

$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed update mbed-os-5.6-oob
$ mbed compile -m NRF52_DK -t ARMC6 -f

**Additional Information**

ARM build (version 5.06) does not have any problem.
ARMC6 build does not seem to startup correctly.  See greentea test result below for more detail.

* ARM build test result 
[NRF52_DK_ARM.txt](https://github.com/ARMmbed/mbed-os/files/1305384/NRF52_DK_ARM.txt)

* ARMC6 build test result
[NRF52_DK_ARMC6.txt](https://github.com/ARMmbed/mbed-os/files/1305386/NRF52_DK_ARMC6.txt)

MarceloSalazar, 2017-09-15:
cc @adbridge

adbridge, 2017-09-15:
@pan- Please take ownership of this one

adbridge, 2017-09-15:
Looks like this may be related https://github.com/ARMmbed/mbed-os/pull/5091

toyowata, 2017-09-20:
Fixed in mbed-os-5.6-oob2.

closed 2017-09-20 by toyowata