ARMmbed/mbed-os #5114: How to split into library and application?

amq, 2017-09-15:
## Question

**How to?**

I'd like to update the application code over OTA without having to transfer the complete binary with mbed-os, mbed-tls, mbed-client etc. Even a complex application has around 40 KB, while the whole package would be 300 KB. This is critical in environments like NB-IoT where the average monthly limit is expected to be 512 KB.

theotherjimmy, 2017-09-15:
So, This is not generally possible. A problem that you will encounter is that as far as the linker is concerned, it may reorder any and all function locations **EVERY TIME YOU COMPILE**. This is not a problem for "regular" applications, but it effectively thwarts your efforts. What you are asking for is not actually to build mbed OS, mbed TLS and others as a library is not what you want. You want to create a separate, possibly bootloader, application and link against that application's copy of mbed OS. This would guarantee that the application will link against the correct location. That's basically incremental link. Mbed OS does not currently support something like this.

0xc0170, 2017-10-13:
@amq Does the above answer your question?

closed 2017-10-13 by amq

amq, 2017-10-13:
@0xc0170 answered (not possible), but I can imagine dynamic linking as a feature in the future. Alternative approach: [binary patching](https://os.mbed.com/blog/entry/towards-fota-lora-crypto-delta-updates/)