ARMmbed/mbed-os #5115: EFM32: Scaling of analogin_read incorrect

seppestas, 2017-09-15:
### Description
- Type: Bug
- Priority: Minor
---------------------------------------------------------------
## Bug

**Target**
EFM32

**Expected behavior**
`AnalogIn::read` should return a value from 0.0 to 1.0 

**Actual behavior**
`AnalogIn::read` returns a value from 0.0 to `0xFFF0/0XFFFF` = 0,99977111467155.

**Cause**
[`analogin_read_u16` returns a value in the range `0x0000 - 0xFFF0` since the resolution of the ADC is 12 bits](https://github.com/ARMmbed/mbed-os/blob/5.4.2/targets/TARGET_Silicon_Labs/TARGET_EFM32/analogin_api.c#L95). However, [in `analogin_read` this value gets divided by `0xFFFF` assuming the range is `0x0000-0xFFFF`](https://github.com/ARMmbed/mbed-os/blob/5.4.2/targets/TARGET_Silicon_Labs/TARGET_EFM32/analogin_api.c#L103). This causes a small error in the value returned by `AnalogIn::read` for the EFM32 target.

0xc0170, 2017-09-21:
cc @stevew817 @akselsm

seppestas, 2017-09-21:
Ow right, I was planning to send a PR for this... Totally forgot XD.

PR #5223 (2017-10-13): Fix analogin scaling for EFM32 target
Fixes #5115.

`analogin_read_u16` returns a value in the range `0x0000 - 0xFFF0`
since the resolution of the ADC is 12 bits. However, in
`analogin_read` this value gets divided by `0xFFFF` assuming the range
is `0x0000-0xFFFF`. This causes a small error in the value returned by
`AnalogIn::read` for the EFM32 target.
commit 81cc5a49 (2017-10-13): Fix analogin scaling for EFM32 target

Fixes #5115.

`analogin_read_u16` returns a value in the range `0x0000 - 0xFFF0`
since the resolution of the ADC is 12 bits. However, in
`analogin_read` this value gets divided by `0xFFFF` assuming the range
is `0x0000-0xFFFF`. This causes a small error in the value returned by
`AnalogIn::read` for the EFM32 target.
closed 2017-10-13 by theotherjimmy