ARMmbed/mbed-os #5117: Too many warnings while building for IAR 7.80 (Debug build)

hasnainvirk, 2017-09-15:
How to reproduce:

Target Platform: MTS_DRAGONFLY_F411RE
Compiler: IAR 7.80
Build type: Debug

There are hundreds of warnings being thrown because of level of optimisation used
for IAR debug build. For example:

 [ 70.9%]: rtx_semaphore.c
[Warning] rtx_semaphore.c@435,4: [Go005]: Cannot inline function "__svcSemaphoreNew" at this level of optimization

hasnainvirk, 2017-09-15:
@MarceloSalazar Opened the ticket here

MarceloSalazar, 2017-09-18:
Thanks, could you please clarify how to reproduce this? (version of mbed OS and example app / test?)

0xc0170, 2017-09-19:
> Thanks, could you please clarify how to reproduce this? (version of mbed OS and example app / test?)

@hasnainvirk Can you paste here or attached the warnings.

The one you shared seems to be generic not target related, however I dont recall seeing one. How to reproduce?

hasnainvirk, 2017-09-19:
@0xc0170 I don't think its target specific. Its just the level of optimization set for debug builds. All the warnings look like the one already in description. It was with mbed-os-5.6 OOB branch.

0xc0170, 2017-09-20:
More information for this issue

https://github.com/ARMmbed/mbed-os/blob/master/tools/profiles/debug.json#L50 - this is the IAR profile, important bit there is `-On`

I assume this is causing problems : https://github.com/ARMmbed/mbed-os/blob/eefb84c0dbd2fe0565fe2dd7888d83775f00cc12/rtos/TARGET_CORTEX/rtx5/core_cm.h#L165 (always inlined attribute).

This is just a warning, but we get it for all SVC calls, so lot of them. I'll talk to IAR and CMSIS teams

0xc0170, 2017-09-22:
I'll report this to CMSIS, and propose a fix. IAR team responded that for now, pragma to  supress the warnings is the way to go.

cmonr, 2018-06-28:
Going to close this since the issue is being tracked here: https://github.com/ARM-software/CMSIS_5/issues/246

closed 2018-06-28 by cmonr