ARMmbed/mbed-os #5118: New mbed-os directory structure doesn't work with mbed-cli

j3270, 2017-09-15:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: https://github.com/ARMmbed/mbed-cli/issues/545
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
ALL, most likely

**Toolchain:**
GCC_ARM|ARM|IAR
same

**Toolchain version:**

**mbed-cli version:** 
mbed-cli version 1.2.0

**mbed-os sha:**
master branch

see the issue posted on mbed-cli for full details

theotherjimmy, 2017-09-15:
Please see the discussion in ARMmbed/mbed-cli#545 for full details of this issue.

0xc0170, 2017-10-13:
@j3270 Is this still relevant? I can see the referenced issue reopened recently?

j3270, 2017-10-13:
@0xc0170 Nope, theotherjimmy fixed it.

Thanks,

closed 2017-10-13 by j3270