ARMmbed/mbed-os #5119: K64F - SPI, AnalogOut, PWM failures (CI test shield)

scartmell-arm, 2017-09-15:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
6

**mbed-cli version:**
1.2.0

**mbed-os sha:**
Current master `https://github.com/ARMmbed/mbed-os/#0906be6397c984026df180b07a3bbd965af52346
`

**Expected behavior**

All tests should pass.

**Actual behavior**
```
+--------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                  | test case                                                              | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-api-analogout         | AnalogOut on AOUT                                                      | 0      | 3      | FAIL   | 0.2                |
| K64F-GCC_ARM | K64F          | tests-api-pwm_rise_fall     | PWM_0 Frequency 100ms                                                  | 0      | 4      | FAIL   | 10.47              |
| K64F-GCC_ARM | K64F          | tests-api-pwm_rise_fall     | PWM_1 Frequency 100ms                                                  | 0      | 4      | FAIL   | 10.48              |
| K64F-GCC_ARM | K64F          | tests-api-pwm_rise_fall     | PWM_2 Frequency 100ms                                                  | 0      | 4      | FAIL   | 10.47              |
| K64F-GCC_ARM | K64F          | tests-api-pwm_rise_fall     | PWM_3 Frequency 100ms                                                  | 0      | 4      | FAIL   | 10.48              |
| K64F-GCC_ARM | K64F          | tests-api-spi               | SPI - SD card exists                                                   | 0      | 1      | FAIL   | 0.09               |
| K64F-GCC_ARM | K64F          | tests-concurrent-comms      | Concurrent testing of I2C and SPI in a single thread                   | 0      | 0      | ERROR  | 0.0                |
| K64F-GCC_ARM | K64F          | tests-concurrent-mixed      | Concurrent testing of Comms and GPIO in a single thread                | 0      | 0      | ERROR  | 0.0                |
+--------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+

mbedgt: test case results: 6 FAIL / 112 OK / 2 ERROR
mbedgt: completed in 624.82 sec
```

**Steps to reproduce**

Run the application at https://github.com/ARMmbed/ci-test-shield

i.e.
```
$ git clone https://github.com/ARMmbed/ci-test-shield.git
$ cd ci-test-shield
```

Set mbed-os version in `mbed-os.lib`

```
$ mbed deploy
$ mbed target auto
$ mbed toolchain GCC_ARM
$ mbed test -n tests-* --app-config .\mbed_app.json
```

0xc0170, 2017-09-19:
cc @mmahadevan108

MateuszMaz, 2018-08-07:
### Description
Hi, I was working on that issue for some time. Here I want to present why this tests failed and how I fixed it.
### tests-api-analogout   
Test is the problem, not api. According to paragraph Customisation in ci-test shield documentation: K64F has not DAC on A5 pin, but on pin which is not connected to CI Test shield. To make a correct test, man should manually fly wire DAC0_OUT to A5 pin on the CI Test shield.

### tests-api-pwm_rise_fall
I resolved this problem. Here is PR: https://github.com/ARMmbed/mbed-os/pull/7716
I think this one needs a little more discussion.
First, the unit of frequency is [hz] not [ms]. Secondly, in all tests we setup pwm by giving period, not frequncy as a parameter. The pwm period we can get at K64F is very limited. This test failed because device was not able to generate pwm with longer period than 69 ms. The solution was increasing the prescaler when we set the period, but we have to remember that if we increase prescaler - we get worse accuracy. My idea was to compute the prescaler in pwmout_period_us function. This way we don't lose accuracy at short pwm periods, and we can get a little greater maximum period value. **I have to mention that without changing the sys clock frequency (what is probably something that we dont want to do), after my fix we are still limited to only 138 ms period, because prescaler which divide the clk, can be set only to following value: 128 (by default it was set to 64 during initialization).** 

### Test suites: tests-concurrent-comms, tests-concurrent-mixed
This problem does not occur in mbed_os version 5.9.4. The reason of error in older versions was too small stack size of Thread_SPI. Update is the solution. 

cc @fkjagodzinski @maciejbocianski @mprse

maclobdell, 2019-01-14:
Thanks everyone who has contributed to this issue.  Essentially, the PWM not being able to pass the 100ms test is the only real issue here.  Subject to NXP approval, I added this as a known issue on the board page https://os.mbed.com/platforms/FRDM-K64F/.  See Important Notes section.  As this issue is extremely old, I'd like to go ahead and close, as it is a documented known issue.  Please re-open if you would like to keep it open.  cc @mmahadevan108

closed 2019-01-14 by maclobdell