ARMmbed/mbed-os #5120: K22F - SPI, AnalogOut, I2C failures (CI test shield)

scartmell-arm, 2017-09-15:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K22F

**Toolchain:**
GCC_ARM

**Toolchain version:**
6

**mbed-cli version:**
1.2.0

**mbed-os sha:**
Current master `https://github.com/ARMmbed/mbed-os/#0906be6397c984026df180b07a3bbd965af52346
`

**Expected behavior**

All tests should pass.

**Actual behavior**
```
+--------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                  | test case                                                              | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| K22F-GCC_ARM | K22F          | tests-api-analogout         | AnalogOut on AOUT                                                      | 0      | 0      | ERROR  | 0.0                |
| K22F-GCC_ARM | K22F          | tests-api-i2c               | tests-api-i2c                                                          | 0      | 1      | ERROR  | 36.41              |
| K22F-GCC_ARM | K22F          | tests-api-spi               | SPI - SD card exists                                                   | 0      | 1      | FAIL   | 1.0                |
| K22F-GCC_ARM | K22F          | tests-concurrent-comms      | tests-concurrent-comms                                                 | 0      | 1      | ERROR  | 59.85              |
| K22F-GCC_ARM | K22F          | tests-concurrent-mixed      | tests-concurrent-mixed                                                 | 0      | 1      | ERROR  | 60.28              |
+--------------+---------------+-----------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 FAIL / 106 OK / 4 ERROR
mbedgt: completed in 686.01 sec

```

**Steps to reproduce**

Run the application at https://github.com/ARMmbed/ci-test-shield

i.e.
```
$ git clone https://github.com/ARMmbed/ci-test-shield.git && cd ci-test-shield
```

Set mbed-os version in `mbed-os.lib`

```
mbed deploy
mbed target auto
mbed toolchain GCC_ARM
mbed test -n tests-* --app-config .\mbed_app.json
```

0xc0170, 2017-09-19:
cc @mmahadevan108

MateuszMaz, 2018-09-20:
### tests-api-analogout
Here we've got wrong test setup. K22F (same as K64F) has not AOUT on pin A5. DAC0_OUT of K22F is not covered by ci_test_shield pins. Additionaly there is no K22F record in mbed_app.json
To perform a valid test, this string should be added to mbed_japp.json

`` 		"K22F": {
            "AOUT": "DAC0_OUT"
        }
``
and DAC0_OUT (look at [schematic](https://os.mbed.com/media/uploads/GregC/frdm-k22f_headers.png)) should be fly wired to A0.

### tests-api-i2c , tests-concurrent-comms, tests-concurrent-mixed
@maciejbocianski 
The important thing here is, **even if I2C api will be changed, it won't pass those tests on K22F.**

Test fails when it attempt to initialize I2C. To be more precise, when PTE0 (i2c_sda) is switched from mode alt3 to alt6 (UART TX to I2C_SDA). Then test returns TIMEOUT, because mbedgt can not recive any message from K22F. 

According to page 13 of [user guide](https://www.nxp.com/docs/en/user-guide/FRDMK22FUG.pdf) 

> The primary serial port interface signals are PTE1 UART1_RX and PTE0 UART1_TX. These signals are connected to the OpenSDAv2 circuit.

I have also checked that if we perform test of i2c initialization, and then switch back PTE0 to UART_TX, then k22f is able to communicate with mbedgt. 

I can suggest some way to perform thoes tests:
We can initialize I2C on PTC11 and PTC10 by adding code below to mbed_app.json

``
		"K22F": {
            "I2C_SDA":"PTC11",
            "I2C_SCL":"PTC10"
        },
``
and fly wire all necessary ports to ci_test_shield + 3,3 V to power sd card reader, eeprom and temperature sensor. This way it works.

MateuszMaz, 2018-09-20:
implementation is ok, additional documentation that describes how to test i2c with greentea framework & ci_test_shield on k22f is needed.