ARMmbed/mbed-os #5121: features-feature_lwip-tests-mbedmicro-net-tcp_hello_world test is failing

studavekar, 2017-09-15:
### Description

As the domain name is changed from developer.mbed.org to os.mbed.com causing network test to fail

```
[1505481840.42][CONN][INF] found KV pair in stream: {{__testcase_name;TCP hello world}}, queued...
[1505481840.46][CONN][RXD] >>> Running case #1: 'TCP hello world'...
[1505481840.51][CONN][INF] found KV pair in stream: {{__testcase_start;TCP hello world}}, queued...
[1505481843.53][CONN][RXD] TCP client IP Address is 10.118.14.63
[1505481843.57][CONN][RXD] HTTP: Connection to developer.mbed.org:80
[1505481843.62][CONN][RXD] HTTP: OK
[1505481843.70][CONN][RXD] :87::FAIL: Expression Evaluated To FALSE
[1505481843.75][CONN][INF] found KV pair in stream: {{__testcase_finish;TCP hello world;0;1}}, queued...
[1505481843.82][CONN][RXD] >>> 'TCP hello world': 0 passed, 1 failed with reason 'Assertion Failed'
[1505481843.82][CONN][RXD]
```
- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
ALL

**Toolchain:**
GCC_ARM|ARM|IAR

geky, 2017-09-15:
The domain was move with an HTTP redirect, here's the response with telnet:
``` http
HTTP/1.1 301 Moved Permanently
Server: nginx/1.11.12
Date: Fri, 15 Sep 2017 16:36:59 GMT
Content-Type: text/html
Content-Length: 186
Connection: close
Location: https://os.mbed.com/
```

Since the test isn't a test of http capability, there's no issues just moving over 👍

theotherjimmy, 2017-09-15:
Sounds great! @geky Are you planning on making a PR to do that? Should we pick it up?

geky, 2017-09-15:
@theotherjimmy, @studavekar was already doing it 😄

theotherjimmy, 2017-09-15:
So it seems!

studavekar, 2017-09-18:
closing it as changes are merged.

closed 2017-09-18 by studavekar