ARMmbed/mbed-os #5126: [OOB 5.6] Blinky failed to build on REALTEK_RTL8195AM (ARMC5, GCC, IAR, online compiler)

neil-tan, 2017-09-16:
### Description
- Type: Bug
- Related issue: 4978
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
ARM
GCC_ARM
IAR

**Toolchain version:**
ARM:  ARM Compiler 5.06 update 5 (build 528)
GCC:  (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release)
IAR: IAR ANSI C/C++ Compiler V7.80.1.11864/W32 for ARM
Online compiler: https://developer-sjc-partnerbeta-border-temporary.mbed.org/compiler/

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
fafc50b

**DAPLink version:**
Updated to the firmware in this page: https://www.amebaiot.com/en/change-dap-firmware/
(also referred by our platform page)
DAP_FW_MBED_RTL8195AM_V0

**Expected behavior**
Blinky should build and the led on the board should pulsate 

**Actual behavior**
ARM: Fails to build  (see ARM.txt)
GCC_ARM: built but no led pulsation observed
IAR: Fails to build. (see IAR.txt)
Online Compiler: Fails to build (see screenshots)

**Steps to reproduce**
mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
git checkout mbed-os-5.6-oob
mbed compile -m REALTEK_RTL8195AM -t [ARM | GCC_ARM | IAR] -f -c

[ARM.txt](https://github.com/ARMmbed/mbed-os/files/1308198/ARM.txt)
[IAR.txt](https://github.com/ARMmbed/mbed-os/files/1308199/IAR.txt)

[image]
[image]

adbridge, 2017-09-18:
So for IAR I get -
```
Link: mbed-os-example-blinky                                                                                                                                                                 
Elf2Bin: mbed-os-example-blinky                                                                                                                                                              
+---------------------------------+--------+-------+------+                                                                                                                                  
| Module                          |  .text | .data | .bss |                                                                                                                                  
+---------------------------------+--------+-------+------+                                                                                                                                  
| [lib]/dl7M_tlf.a                |   4224 |     0 |   40 |                                                                                                                                  
| [lib]/lib_peripheral_mbed_iar.a |   8802 |     0 |  480 |                                                                                                                                  
| [lib]/librom.a                  | 276520 |     0 |    0 |                                                                                                                                  
| [lib]/m7M_tl.a                  |   2520 |     0 |    0 |                                                                                                                                  
| [lib]/rt7M_tl.a                 |    936 |     0 |    0 |                                                                                                                                  
| [lib]/th7M_tlf.a                |    192 |     0 |   16 |                                                                                                                                  
| [misc]                          |    428 |     0 |    0 |                                                                                                                                  
| main.o                          |     64 |     0 |   20 |                                                                                                                                  
| mbed-os/hal                     |    216 |     0 |    0 |                                                                                                                                  
| mbed-os/platform                |    558 |     0 |   17 |                                                                                                                                  
| mbed-os/rtos                    |   8156 |     0 | 6344 |                                                                                                                                  
| mbed-os/targets                 |   1676 |     0 | 2836 |                                                                                                                                  
| Subtotals                       | 304292 |     0 | 9753 |                                                                                                                                  
+---------------------------------+--------+-------+------+                                                                                                                                  
Total Static RAM memory (data + bss): 9753 bytes                                                                                                                                             
Total Flash memory (text + data): 304292 bytes                                                                                                                                               
                                                                                                                                                                                             
Image: .\BUILD\REALTEK_RTL8195AM\IAR\mbed-os-example-blinky.bin

adbridge, 2017-09-18:
For the online IDE the actual error appears to be - 
Error: "/extras/mbed-os.lib/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/rtl8195a.sct", line 16 (column 22): Error: L6236E: No section matches selector - no section to be FIRST/LAST.
Plus a huge number of warnings!

adbridge, 2017-09-18:
@Archcady Is https://github.com/ARMmbed/mbed-os/pull/5044 likely to fix any of the above or is this a different issue ?

Archcady, 2017-09-19:
HI @adbridge , that's a different issue when compile with debug.json at mbed-os/tools/profiles

0xc0170, 2017-09-19:
>that's a different issue when compile with debug.json at mbed-os/tools/profiles

@Archcady Are you able to reproduce this issue?

Archcady, 2017-09-20:
Give me sometime to test, but please notice that no LED on board will pulsate since we define LED1 in pinnames.h to be PB_4/D8. The three leds on board are reserved for other usage.

MarceloSalazar, 2017-09-23:
Confirmed fixed in 5.6 rc2.

closed 2017-09-23 by MarceloSalazar