ARMmbed/mbed-os #5127: test command scans and builds unnecessary components

toyowata, 2017-09-16:
Moved from [here](https://github.com/ARMmbed/mbed-cli/issues/546).

### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug
Mbed CLI scans and builds unnecessary components which are located to one upper level of current directory.  This causes longer build time.

**Target**
All

**Toolchain:**
All

**mbed-cli version:** 
1.2.0

**Expected behavior**
Mbed CLI scan and build source code in current directory.

**Actual behavior**
Mbed CLI scan and build source code in current directory and one upper level of current directory.  An example build result is below:

```
$ ls

BLEClient_mbedDevConn				mbed-os-example-ble
BLE_Server_BME280				mbed-os-example-blinky
MB85RSxx_Hello					mbed-os-example-client
MB85RSxx_TEST					mbed-os-example-tls
esp8266_v2.0_firmware				mbed-os-example-wifi
esptool						mbed_blinky
esptool.zip					pyOCD
manifest-tool-internal				security
wifi55						mbed-os

$ cd mbed-os
$ mbed test --compile -m K64F -t GCC_ARM

Building library mbed-build (K64F, GCC_ARM)
Scan: mbed-os
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Copy: mbed.h

(snip)

Compile [100.0%]: fsl_common.c
Scan: GCC_ARM
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Building project div (K64F, GCC_ARM)
Scan: GCC_ARM
Scan: div
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Compile [100.0%]: main.cpp
Link: div
Elf2Bin: div
+--------------------------------+-------+-------+------+
| Module                         | .text | .data | .bss |
+--------------------------------+-------+-------+------+
| BLEClient_mbedDevConn/BUILD    |   895 |     0 |    8 |
| BLE_Server_BME280/BUILD        |  3938 |   176 | 4364 |
| MB85RSxx_Hello/mbed            |  3650 |     8 |  112 |
| MB85RSxx_TEST/mbed             |  1579 |     4 |   36 |
| [fill]                         |   145 |     4 | 2252 |
| [lib]/libc.a                   | 31093 |  2472 |   56 |
| [lib]/libgcc.a                 |  3128 |     0 |    0 |
| [lib]/libnosys.a               |    32 |     0 |    0 |
| [lib]/misc                     |   232 |     8 |   28 |
| mbed-os-example-blinky/BUILD   |  2449 |     0 |  488 |
| mbed-os-example-client/BUILD   |  1917 |    36 |    9 |
| mbed-os-example-tls/tls-client |  2270 |     0 | 1644 |
| mbed-os-example-wifi/BUILD     |   540 |     0 |    5 |
| mbed-os/BUILD                  |   795 |     4 |    2 |
| mbed_blinky/mbed               |  4634 |     4 |  520 |
| wifi55/BUILD                   |   396 |     0 |    0 |
| Subtotals                      | 57693 |  2716 | 9524 |
+--------------------------------+-------+-------+------+
Total Static RAM memory (data + bss): 12240 bytes
Total Flash memory (text + data): 60409 bytes
Image: BUILD/tests/K64F/GCC_ARM/TESTS/mbedmicro-mbed/div/div.bin

```

**Steps to reproduce**

$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed compile -m K64F -t GCC_ARM
$ cd ..
$ mbed import mbed-os
$ cd mbed-os
$ mbed test --compile -m K64F -t GCC_ARM

0xc0170, 2017-09-19:
cc @theotherjimmy

theotherjimmy, 2017-09-19:
Resolved by #5125

PR #5125 (2017-09-21): Refactor memap for speed
Improves performance by 10-15x on my machine. See below for individual compiler comparisons.

Resolves #4976
Resolves #5124
Resolves #5127 

Todo:
 - [x] Parsing tests (let's never let this non-performance bug in again)
 - [x] Resolve the `Malformed input...` output from GCC
closed 2017-09-21 by theotherjimmy