ARMmbed/mbed-os #5128: NUCLEO_L476RG led1.read() shows wrong value

andrewc-arm, 2017-09-18:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

---------------------------------------------------------------

Hi,

NUCLEO_L476RG led1.read() shows wrong value.
This is the blinky code with logs.
```
#include "mbed.h"

DigitalOut led1(LED1);

// main() runs in its own thread in the OS
int main() {
    while (true) {
        led1 = !led1;
        printf("Changed the LED(%d)\n", led1.read());
        wait(0.5);
    }
}
```

This is the output.

```
Changed the LED(1)
Changed the LED(1)
Changed the LED(1)
```

The expected output is following.
```
Changed the LED(0)
Changed the LED(1)
Changed the LED(0)
Changed the LED(1)
```

I tested with following mbed-os-example-blinky version but failed.
```
commit 12b6eaeea61c2887dccf1e05c45035b8e545e134
Merge: 83bb825 6d96d2f
Author: Martin Kojtal <0xc0170@gmail.com>
Date:   Fri Sep 15 07:32:34 2017 +0100

    Merge pull request #82 from adbridge/master

    Updating mbed-os to mbed-os-5.5.7
```

Then I changed mbed-os to mbed-os-5.6 branch and the result is same.

Thanks,

jeromecoutant, 2017-09-18:
Hi
I am sorry, but I can't reproduce your issue...
Jerome

andrewc-arm, 2017-09-21:
Interesting. There must be some difference between your setting and my setting, then. To double check, could you kindly post the actual image you tested? I will apply to my board and make sure it's setting issue.

0xc0170, 2017-10-13:
> Interesting. There must be some difference between your setting and my setting, then. To double check, could you kindly post the actual image you tested? I will apply to my board and make sure it's setting issue.

If you can share what board you have (any revision)? More details so we can compare what differences are there.

bcostm, 2017-10-16:
Same for me. Your sample works perfectly on my board.

Here my bin file:
[blink_test.zip](https://github.com/ARMmbed/mbed-os/files/1387327/blink_test.zip)

andrewc-arm, 2017-10-17:
Hi, I checked blink_test.zip works to my board. Then it must be some local setup issue. I will investigate later. Thanks!

closed 2017-10-17 by andrewc-arm