ARMmbed/mbed-os #5129: UBLOX_EVK_ODIN_W2 failing to blink D2 and D8

andrewc-arm, 2017-09-18:
### Description
- Type: Bug
- Related issue: `#5128`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM
Please note that hackathon developers can only use GCC_ARM.

**Toolchain version:**
5.4.1 20160919

**mbed-cli version:** 
(`mbed --version`)
1.2.0

**mbed-os sha:**
(`git log -n1 --oneline`)
mbed-os-5.6
a4056fb Added versionning define block for new minor release.

**DAPLink version:**

**Expected behavior**
These code should work well.
```
DigitalOut led1(D2);
DigitalOut led2(D8);
```

**Actual behavior**
led1 and led2 above do not work and blink properly.

**Steps to reproduce**
* Get UBLOX_EVK_ODIN_W2
* Get mbed-os-example-blinky code.
* Make small LED + resistor network with D2 pin.
* Run following code.
```
#include "mbed.h"

DigitalOut led1(D2);

// main() runs in its own thread in the OS
int main() {
    while (true) {
        led1 = !led1;
        printf("Changed the LED(%d)\n", led1.read());
        wait(0.5);
    }
}
```
* You will observe no led1 blinking.
* Change to D0 and it will work.

----------------------------------------------------------------

Hi,

This is quite serious. For Seoul IoT Hackathon (http://www.seoulhackathon.org/), we picked UBLOX_EVK_ODIN_W2 board and D2 and D8 pins are not working. Please investigate this issue and give us solution.

Thanks,

janjongboom, 2017-09-18:
D2/D8 are used for UART1 (like many ST boards). Is this UART used?

Maybe shared with USBTX/USBRX or used for another peripheral?

sarahmarshy, 2017-09-18:
[D2](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_UBLOX_EVK_ODIN_W2/PinNames.h#L138) and [D8](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_UBLOX_EVK_ODIN_W2/PinNames.h#L144) are used for USBTX/USBRX on this board, so they've already been initialized for that use.

andrewc-arm, 2017-09-19:
Hi,

Thanks for the feedback. I am a bit confused because I tried this code from K64F and got every GPIO working for D0 ~ D15.
```
#include "mbed.h"

DigitalOut led00(D0);
DigitalOut led01(D1);
DigitalOut led02(D2);
DigitalOut led03(D3);
DigitalOut led04(D4);
DigitalOut led05(D5);
DigitalOut led06(D6);
DigitalOut led07(D7);
DigitalOut led08(D8);
DigitalOut led09(D9);
DigitalOut led10(D10);
DigitalOut led11(D11);
DigitalOut led12(D12);
DigitalOut led13(D13);
DigitalOut led14(D14);
DigitalOut led15(D15);

// main() runs in its own thread in the OS
int main() {
    while (true) {
        led00 = !led00;
        led01 = !led01;
        led02 = !led02;
        led03 = !led03;
        led04 = !led04;
        led05 = !led05;
        led06 = !led06;
        led07 = !led07;
        led08 = !led08;
        led09 = !led09;
        led10 = !led10;
        led11 = !led11;
        led12 = !led12;
        led13 = !led13;
        led14 = !led14;
        led15 = !led15;
		printf("Changed the LEDs(%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d)\n", 
				led00.read()
				,led01.read()
				,led02.read()
				,led03.read()
				,led04.read()
				,led05.read()
				,led06.read()
				,led07.read()
				,led08.read()
				,led09.read()
				,led10.read()
				,led11.read()
				,led12.read()
				,led13.read()
				,led14.read()
				,led15.read()
			  );
        wait(0.5);
    }
}
```

So, I checked u-blox Mbed page and found following.
https://os.mbed.com/platforms/ublox-EVK-ODIN-W2/
```
By default pins A1 and A2 will be configured as digital I/Os. To use them as analog inputs, configure the jumpers on pin header J26.
```
However, I could not observe D2 or D8 mentioned in the page. Could you double check if it is (D2,D8) or (A1,A2) for USB serial?

Do you still think the developers cannot use D2 or D8 for EVK-ODIN-W2? Do you think J26 jumper or any other jumper can solve this problem? I thought DAPLink's serial is not related with board's digital outputs.

Thanks,

andrewc-arm, 2017-09-19:
Hi,

Just one more question. I changed the J13 to connect 1-3 and 2-4 as shown in the schematic but I don't still enable DigitalOut of D2 and D8.
[image]

Maybe there is some hard coded UART1 (D2, D8) initialized from the driver?
Could you please dig into it and help the developers to utilize D2 and D8 pins?

Thanks,

jeromecoutant, 2017-09-19:
> Maybe there is some hard coded UART1 (D2, D8) initialized from the driver?

Maybe check PeripheralNames.h TARGET_UBLOX_EVK_ODIN_W2 file:
#define STDIO_UART_TX  PA_9
#define STDIO_UART_RX  PA_10

and PinNames.h :
    D8      = PA_9,   // UART1-TX
    D2      = PA_10,  // UART1-RX

This serial init is done in platform/mbed_retarget.cpp :

#if DEVICE_SERIAL
    serial_init(&stdio_uart, STDIO_UART_TX, STDIO_UART_RX);

Regards,

0xc0170, 2017-09-19:
> Could you please dig into it and help the developers to utilize D2 and D8 pins?

Can you please, @andreaslarssonublox @andreaspeterssonublox ?

andrewc-arm, 2017-09-21:
Hello, u-blox team, @andreaslarssonublox and @andreaspeterssonublox 

Is there any progress to this?

I did some research and it matched @jeromecoutant and other engineers' analysis. 
I think there should be some configuration or simple definition to toggle (D2,D8) and (D0,D1). (If needed, we can do extra jumper setting.) So that the developers can have a bypass to use whichever pin they want to use. At current stage, (D2,D8) pair is solely occupied. If so, I think we should mention it from the board page. Without this mentioned, it will confuse developers using this board.

andreaslarssonublox, 2017-09-21:
Thanks for the explanation @jeromecoutant. We will fix the hardcoded pins in PeripheralNames.h so it uses USBTX and USBRX from pinnames.h defines instead.

However I'm not sure how to redefine the pinnames in mbed_app.json. Do you know @0xc0170 ?

0xc0170, 2017-09-21:
>However I'm not sure how to redefine the pinnames in mbed_app.json. Do you know @0xc0170 ?

Not sure what should be the outcome. Can you share the pseudo-code? I would assume that something like some ST targets have, that based on the target's configuration, they redefine some pins (target's configuration would set the default). 

(pseudocode)`USBTX` would be set as `MBED_CONF_USBX`, and this could be overwritten by an app. Is this what you are looking for?

andreaslarssonublox, 2017-09-21:
Thanks @0xc0170!
That is probably the best way to do it unless there is some more mbedish way of doing it.

Usage in mbed_app.json:
    "config": {
        "USBRX":"D0",
        "USBTX":"D1",

peripheralnames.h:

#ifdef MBED_CONF_APP_USBTX
#define STDIO_UART_TX  MBED_CONF_APP_USBTX
#else
#define STDIO_UART_TX  USBTX
#endif

#ifdef MBED_CONF_APP_USBRX
#define STDIO_UART_RX  MBED_CONF_APP_USBRX
#else
#define STDIO_UART_RX  USBRX
#endif

0xc0170, 2017-09-21:
One note, there is target config and app config. I believe this is target related, thus would end up in target configuration, and can be changed in the app, by target overrides. Thus would be `MBED_CONF_TARGETxxxx`.

andreaslarssonublox, 2017-09-25:
@0xc0170 I have created a PR #5192. Can you also check that this is the preferred way to do it?

andrewc-arm, 2017-09-27:
Thanks for enabling this, @andreaslarssonublox 
Due to being busy, I was not able to test PR #5192. I will do so later.

0xc0170, 2017-10-13:
@andrewc-arm If you find any issues, please reopen. The fix was integrated

closed 2017-10-13 by 0xc0170