ARMmbed/mbed-os #5131: make_gcc_arm exporter broken

SeppoTakalo, 2017-09-18:
### Description
- Type: Bug
- Priority: Blocker

When trying to export `mbed-os-example-mesh-minimal` the exporter fails

```
~/src/mbed-os-example-mesh-minimal$ mbed export -m K64F -i make_gcc_arm
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_NANOSTACK
Scan: FEATURE_STORAGE
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_COMMON_PAL
Traceback (most recent call last):
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/project.py", line 256, in <module>
    main()
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/project.py", line 252, in main
    build_profile=profile, app_config=options.app_config)
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/project.py", line 95, in export
    app_config=app_config)
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/export/__init__.py", line 341, in export_project
    macros=macros)
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/export/__init__.py", line 207, in generate_project_files
    exporter.generate()
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/export/makefile/__init__.py", line 120, in generate
    ctx.update(self.format_flags())
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/export/makefile/__init__.py", line 140, in format_flags
    for k, v in self.flags.iteritems():
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/export/exporters.py", line 111, in flags
    for_asm=True)
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/toolchains/gcc.py", line 161, in get_compile_options
    opts += ['@%s' % self.get_inc_file(includes)]
  File "/Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/toolchains/__init__.py", line 831, in get_inc_file
    include_file = join(self.build_dir, ".includes_%s.txt" % self.inc_md5)
AttributeError: 'GCC_ARM' object has no attribute 'inc_md5'
[mbed] ERROR: "/usr/local/opt/python/bin/python2.7" returned error code 1.
[mbed] ERROR: Command "/usr/local/opt/python/bin/python2.7 -u /Users/septak01/src/mbed-os-example-mesh-minimal/mbed-os/tools/project.py -i make_gcc_arm -m K64F --source ." in "/Users/septak01/src/mbed-os-example-mesh-minimal"
---
```

Exporting the blinky seems to work.

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
`1.1.1`

**mbed-os sha:**
5.6-rc1

theotherjimmy, 2017-09-18:
@SeppoTakalo I'm closing this as it's a duplicate of issue #5092, which was resolved by PR #5093 .

theotherjimmy, 2017-09-18:
This regression has already been fixed on master. Please update your copy of mbed-os.

closed 2017-09-18 by theotherjimmy